righthalfplane/SdrGlut #3: SDR File Format?

dbnut, 2020-01-22:
Hi, Dale.
Please explain what file formats SdrGlut can read?
Paul

righthalfplane, 2020-01-23:
SdrGlut reads/writes 4 byte floating point in pairs with the real followed by the imaginary. It is in a single array declared by a (float *) statement in C. The file names have the center frequency and the sample rate.

dbnut, 2020-01-23:
Thanks for that. Sorry, little-endian? I'll try this soon.

righthalfplane, 2020-01-23:
This video  shows the saving and the loading of the I/Q files -

https://www.youtube.com/watch?v=zxgcRjTahcA

The program uses whatever the machine it is running on uses - they are almost all little-endian.
When the Macintosh was big-endian, I made my programs do both, but now it is little-endian - so I do not bother anymore.

dbnut, 2020-01-23:
Mmm, I skipped too fast through that video first time around. Nice job. I'll have to see if some other RAW files will play back in yours, and whether my GNYRadio flow can handle SdrGlut files!
Just a suggestion...
Playback timeline, set marks A & B, keep replaying that segment to catch an ID or whatever.
Will close now. Many thanks, Dale.

closed 2020-01-23 by dbnut