righthalfplane/SdrGlut #5: Error trying to run the SdrGlut-521-uBuntu18.04.zip version

MaxQ73, 2020-03-19:
Getting this error message when trying to run this version. I'm using Linux Mint 19.1.

./sdrglut.x: error while loading shared libraries: librtaudio.so.6: cannot open shared object file: No such file or directory

I don't see that file in the file list. There is librtaudio.so.6.1.0.

righthalfplane, 2020-03-19:
librtaudio.so.6 is in my copy of  SdrGlut-521-uBuntu18.04.zip - so it should be yours.

You need to start sdrglut.x by running the script -

runSdrGlut.sh

It setup up sets up the environment needed to run the program.
Running sdrglut.x directly will not work.

MaxQ73, 2020-03-19:
It failed when I tried to run the script, so I ran the commands in the script individually in a terminal in order to see the error message.

I renamed librtaudio.so.6.1.0 to librtaudio.so.6 and it is running now.

closed 2020-03-25 by MaxQ73