righthalfplane/SdrGlut #7: Question regarding Direct Sample Mode

Brent-Davis, 2020-03-29:
Just wanted to say I'm loving this app but I can't figure out how to set Direct Sampling mode when using this with an RTL_SDR v3. In most other applications like GQRX its just a matter of setting an argument. Are there any arguments that are passable for SdrGlut?

righthalfplane, 2020-04-01:
I am not sure how set things up for Direct Sampling mode - I will have to look into it.

righthalfplane, 2020-04-02:
That was simple enough - it will be in the next release. If you are in a hurry, you can clone the SdrGlut directory and build SdrGlut - the Direct Sampling mode is now in a pop up menu.

Brent-Davis, 2020-04-02:
Thank you so much for working on this. It would be really awesome to get on HF with the RTLSDR v3 dongle. I just cloned the SdrGlut directory and tried to compile but I'm currently getting an error. Trying to poke through myself and see if I can find a solution but I'm not much of a coder. Anyway here's what I am getting:

Radio/Radio.cpp: In member function ‘virtual int Radio::OpenWindows(Scene*)’:
Radio/Radio.cpp:1078:16: warning: converting to non-pointer type ‘int’ from NULL [-Wconversion-null]
     int menu62=NULL;
                ^~~~
g++ -O3  -std=gnu++11 -DNO_FORTRAN -pthread -I./glui-routines/include -I/usr/local/hdf-64/include -I/usr/X11R6/include  -I./DrawFindPointHistory  -I ./SdrFile -I./MainSource -I./KdTree-routines -I./glui-routines/include -I./CExpress -I./Classes -I./SdrGlut -I./FMRadio  -I./Radio -Wall    -c -o MainSource/fir.o MainSource/fir.cpp
Radio/RaceFastsdr.cpp: In function ‘int ProcessSound(void*)’:
Radio/RaceFastsdr.cpp:322:12: warning: variable ‘start’ set but not used [-Wunused-but-set-variable]
     double start=rtime();
            ^~~~~
Radio/RaceFastsdr.cpp: In function ‘int setFilters(playData*, Filters*)’:
Radio/RaceFastsdr.cpp:780:50: error: invalid conversion from ‘int’ to ‘liquid_ampmodem_type’ [-fpermissive]
     f->demodAM = ampmodem_create(0.5, mode, iflag);
                                                  ^
Radio/RaceFastsdr.cpp:780:50: error: too few arguments to function ‘ampmodem_s* ampmodem_create(float, float, liquid_ampmodem_type, int)’
In file included from Radio/RaceFastsdr.h:11:0,
                 from Radio/RaceFastsdr.cpp:3:
/usr/include/liquid/liquid.h:5906:10: note: declared here
 ampmodem ampmodem_create(float _m,
          ^~~~~~~~~~~~~~~
SdrFile/SdrFile.cpp: In function ‘int setFilters(playData4*, Filters2*)’:
SdrFile/SdrFile.cpp:842:50: error: invalid conversion from ‘int’ to ‘liquid_ampmodem_type’ [-fpermissive]
     f->demodAM = ampmodem_create(0.5, mode, iflag);
                                                  ^
SdrFile/SdrFile.cpp:842:50: error: too few arguments to function ‘ampmodem_s* ampmodem_create(float, float, liquid_ampmodem_type, int)’
In file included from SdrFile/SdrFile.h:19:0,
                 from SdrFile/SdrFile.cpp:11:
/usr/include/liquid/liquid.h:5906:10: note: declared here
 ampmodem ampmodem_create(float _m,
          ^~~~~~~~~~~~~~~

Brent-Davis, 2020-04-02:
Got it to compile! These lines were causing the issue for me:
#else
    f->demodAM = ampmodem_create(0.5, mode, iflag);
Once I changed those I got rid of the errors and it compiled. And direct sampling is working great. Listening to HF right now as we speak. Thank you so much for adding it!

closed 2020-04-04 by Brent-Davis