righthalfplane/SdrGlut #10: downloaded macOS executable v543 fails to start

bethanysciences, 2020-05-07:
downloaded macOS executable SdrGlut-543-Install.dmg, unpacked and moved SdrGlut.app to applications.  
  
Executing opens terminal for a sec then closes it.  
  
SdrGlut.app package file contents  
  
"SdrGlut"  
```  
#!/bin/bash
export DYLD_LIBRARY_PATH=.
export SOAPY_SDR_ROOT=.
#export DYLD_PRINT_LIBRARIES=1
cd `dirname $0`
open -a Terminal.app SdrGlut3.sh
```  

"SdrGlut3.sh"  
```  
#!/bin/bash
export DYLD_LIBRARY_PATH=.
export SOAPY_SDR_ROOT=.
#export DYLD_PRINT_LIBRARIES=1
cd `dirname $0`
./SdrGlut2
```

righthalfplane, 2020-05-08:
If you go into the package and try to run it - what happens -

cd /Users/dir/Desktop/SdrGlut.app/Contents/MacOS
export DYLD_LIBRARY_PATH=.
export SOAPY_SDR_ROOT=.
export DYLD_PRINT_LIBRARIES=1
./SdrGlut2

bethanysciences, 2020-05-08:
Here is what I get:

bobbysmith@mbp16 MacOS % export DYLD_LIBRARY_PATH=.
bobbysmith@mbp16 MacOS % export SOAPY_SDR_ROOT=.
bobbysmith@mbp16 MacOS % export DYLD_PRINT_LIBRARIES=1
bobbysmith@mbp16 MacOS % ls
SdrGlut				libfftw3f.3.dylib
SdrGlut2			libhackrf.0.dylib
SdrGlut3.sh			libhamlib.2.dylib
lib				libliquid.dylib
libLimeSuite.19.04-1.dylib	librtaudio.6.dylib
libSoapySDR.0.7.dylib		librtlsdr.0.dylib
libairspy.0.dylib		libusb-1.0.0.dylib
libairspyhf.0.dylib
bobbysmith@mbp16 MacOS % ./SdrGlut2
dyld: loaded: <97EDC608-1871-30E5-A295-1A4B4A201D84> /Users/bobbysmith/Desktop/SdrGlut.app/Contents/MacOS/./SdrGlut2
dyld: loaded: <4E9C4B23-6D44-3804-AFF8-84C3B060E8F5> /System/Library/Frameworks/OpenGL.framework/Versions/A/OpenGL
dyld: loaded: <AA7DDB1D-CAE5-3884-A129-35C5FDF99BA4> ./librtaudio.6.dylib
dyld: loaded: <3BB4E75E-557A-39D6-BBE7-4416F28CAC4E> /System/Library/Frameworks/OpenAL.framework/Versions/A/OpenAL
dyld: loaded: <3F0E363C-9380-3226-A4D1-67E740079AAD> /System/Library/Frameworks/OpenCL.framework/Versions/A/OpenCL
dyld: loaded: <F2A60855-A30A-3128-914A-BB3B50D39431> ./libliquid.dylib
dyld: loaded: <847A423D-1FF7-3BCC-BF1E-13DECDAA9E3D> ./libSoapySDR.0.7.dylib
dyld: Library not loaded: /opt/X11/lib/libglut.3.dylib
  Referenced from: /Users/bobbysmith/Desktop/SdrGlut.app/Contents/MacOS/./SdrGlut2
  Reason: image not found
zsh: abort      ./SdrGlut2
bobbysmith@mbp16 MacOS % 

looks like x11 issue again.

Do I have to use XQuartz? 

Or, in your directions:  "To compile with the GUI envoke - SdrGlut.xcodeproj"

not quite sure what this means - would I be able to build this in xcode and bypass the need for x11?

Best,

Bobby

---
Robert Smith II
www.rpsmith.net
+1 404-583-8114
rpsmithii@mac.com

righthalfplane, 2020-05-08:
I forgot that Xquartz needed to be loaded. It has been on my system for years. That loads the X-windows stuff into /opt/X11

https://www.xquartz.org/releases/

righthalfplane, 2020-05-08:
"To compile with the GUI envoke - SdrGlut.xcodeproj"

Double click on the Xcode project "SdrGlut.xcodeproj" will bring up Xcode to compile the project - assuming the you have Xcode 9.2 or newer installed on your system.

closed 2020-05-11 by righthalfplane

bethanysciences, 2020-05-13:
Thanks - works as it should now.