righthalfplane/SdrGlut #11: Fails to run

davidswelt, 2020-06-03:
I'm running macOS 10.15.4

It opens a Terminal window (not really appropriate for a Mac app in the first place), which shows:

Last login: Sun May 31 15:52:13 on ttys004
dr@janelle ~ % /var/folders/1v/ybw447ls4zs_9_v4s8gszqx80000gp/T/AppTranslocation/69D15E3A-801B-4963-9549-15E7A25EE2B3/d/SdrGlut.app/Contents/MacOS/SdrGlut3.sh ; exit;
zsh: no such file or directory: /var/folders/1v/ybw447ls4zs_9_v4s8gszqx80000gp/T/AppTranslocation/69D15E3A-801B-4963-9549-15E7A25EE2B3/d/SdrGlut.app/Contents/MacOS/SdrGlut3.sh

righthalfplane, 2020-06-03:
What does it do when you try to run it from the desktop or applications ?

If that does not work -
Go into the app package  folder  "Contents/MacOS" and enter -

export DYLD_LIBRARY_PATH=.
export SOAPY_SDR_ROOT=.
SdrGlut2

Show me the output  ?

davidswelt, 2020-06-04:
When copied somewhere else, I still get the extra window showing what's
pasted below.

I do get a XQuartz window with a menu, but the actual menu at the top of
the screen doesn't have application-specific entries.

Last login: Tue Jun  2 20:52:14 on ttys000

/Users/dr/Dropbox/MacSync/Desktop/SdrGlut.app/Contents/MacOS/SdrGlut3.sh ;
exit;

dr@janelle ~ %
/Users/dr/Dropbox/MacSync/Desktop/SdrGlut.app/Contents/MacOS/SdrGlut3.sh ;
exit;

Lib Version: v0.7.0-g69c16e98

API Version: v0.7.0

ABI Version: v0.7

Install root: .

Search path: ./lib/SoapySDR/modules0.7

Module found: ./lib/SoapySDR/modules0.7/libHackRFSupport.so

Module found: ./lib/SoapySDR/modules0.7/libIrisSupport.so

Module found: ./lib/SoapySDR/modules0.7/libLMS7Support.so

Module found: ./lib/SoapySDR/modules0.7/libairspySupport.so

Module found: ./lib/SoapySDR/modules0.7/libairspyhfSupport.so

Module found: ./lib/SoapySDR/modules0.7/libnetSDRSupport.so

Module found: ./lib/SoapySDR/modules0.7/libremoteSupport.so

Module found: ./lib/SoapySDR/modules0.7/librtlsdrSupport.so

Module found: ./lib/SoapySDR/modules0.7/libsdrPlaySupport.so

  dlopen() failed: dlopen(./lib/SoapySDR/modules0.7/libsdrPlaySupport.so,
1): Library not loaded: libmirsdrapi-rsp.so.2.13

  Referenced from:
/Users/dr/Dropbox/MacSync/Desktop/SdrGlut.app/Contents/MacOS/lib/SoapySDR/modules0.7/libsdrPlaySupport.so

  Reason: image not found

davidswelt, 2020-06-04:
I tried starting from the Terminal as you suggested with the shell
variables reset, and the error messages are exactly the same.

righthalfplane, 2020-06-04:
Everything looks Ok - unless you are trying to use a sdrplay device. SDRplay_RSP_API-MacOSX-2.13.2 needs to be loaded for that. What device are you trying to use it with.

davidswelt, 2020-06-04:
The RTL2832U, and it's recognized and working.

So, this bug is mainly that you're can't run from the DMG. This seems
pertinent: https://www.synack.com/blog/untranslocating-apps/

righthalfplane, 2020-06-04:
Your right - translocating is some thing new - I was wondering how you got things at -
 /var/folders/1v/ybw447ls4zs_9_v4s8gszqx80000gp/T/AppTranslocation/69D15E3A-801B-4963-9549-15E7A25EE2B3/d/SdrGlut.app/Contents/MacOS/SdrGlut3.sh

closed 2020-06-18 by righthalfplane