righthalfplane/SdrGlut #12: Frequency offset for up / down converters

nmaster2042, 2020-07-04:
I'm using an up converter with my sdr for HF.

When I start Sdrglut, I chose a sdr and adjust frequency and sample rate.

I didn't find a way to enter an offset so I can read actual frequency when using up converter.

This would be useful.

righthalfplane, 2020-07-04:
At first I thought that it was going to be difficult to do, but later I realized it would be easy. It will be in the next release version or If you want to rebuild from the source, you can get it now. The "Frequency Offset" is now an option in the SDR Dialog menu.

nmaster2042, 2020-07-04:
I tried the offset, It doesn't works as I exptected. I think I have not well explained the purpose of my request, sorry.
I have an up converter that has an offset of 125 mhz.
On the offset option I set -125000000, but on software it doesn't change frequency I read on screen.
On software side, up converter on, I want to tune to 10 Mhz, so I enter 10000000 on frequency and center frequency.
And can read 10 mhz on graph window.
On hardware side, considering the offset, to tune to 10 mhz, the actual frequency tuned would be 135 mhz.
Here is the same use case but on gqrx: http://www.hagensieker.com/blog/page/?post_id=68&title=how-to-use-an-upconverter-with-sdr-in-gqrx

righthalfplane, 2020-07-04:
Set to offset to +125000000 should give you what you want.

nmaster2042, 2020-07-04:
OK. It's reversed compared to other sdr softwares, so I couldn't tune correctly.
But it's not a real issue, just have to remember it.
Will make more tests asap.
Thank you very much.

righthalfplane, 2020-07-05:
I flipped offset - I guess it makes a like more sense that way. Use  -125e6 with the latest version.

nmaster2042, 2020-07-05:
I tried again, all is fine now, -125e6 as offset gives expected result.

closed 2020-07-05 by righthalfplane