righthalfplane/SdrGlut #13: SDRGLUT on Big Sur

yann44, 2020-11-17:
Hi,
Thanks for this good software which worked very well with MacOS Catalina.
My computer changed his OS and now it's running MacOS Big Sur.
I did the same install on Big Sur to use it (Xquartz, Xcode, RTAudio and SdrGlut 7.2.1) but, this time, the software doesn't start.
Here is the Terminal view:

/Applications/SdrGlut.app/Contents/MacOS/SdrGlut3.sh ; exit;                    
yann@iMac-de-yann ~ % /Applications/SdrGlut.app/Contents/MacOS/SdrGlut3.sh ; exit;
Lib Version: v0.7.0-g69c16e98
API Version: v0.7.0
ABI Version: v0.7
Install root: .
Search path: ./lib/SoapySDR/modules0.7
Module found: ./lib/SoapySDR/modules0.7/libHackRFSupport.so
Module found: ./lib/SoapySDR/modules0.7/libIrisSupport.so
Module found: ./lib/SoapySDR/modules0.7/libLMS7Support.so
Module found: ./lib/SoapySDR/modules0.7/libairspySupport.so
Module found: ./lib/SoapySDR/modules0.7/libairspyhfSupport.so
Module found: ./lib/SoapySDR/modules0.7/libnetSDRSupport.so
Module found: ./lib/SoapySDR/modules0.7/libremoteSupport.so
Module found: ./lib/SoapySDR/modules0.7/librtlsdrSupport.so
Module found: ./lib/SoapySDR/modules0.7/libsdrPlaySupport.so
freeglut (./SdrGlut2): failed to open display ':0'
Saving session...
...copying shared history...
...saving history...truncating history files...
...completed.
Deleting expired sessions...none found.

[Opération terminée]

Do you think it is possible to fix the problem?

Thanks in advance,

Yann

righthalfplane, 2020-11-17:
Do glxgears and xclock work ?

yann44, 2020-11-17:
Same problem:

yann@MacBook-Air-de-yann ~ % glxgears
Error: couldn't open display :0
yann@MacBook-Air-de-yann ~ % xclock
Error: Can't open display: :0

yann44, 2020-11-17:
Problem solved !!!

In fact, XQuartz MUST run before SdrGlut launch...

And it works very well with my HackRf on macOS Big Sur !!!!

Thanks a lot.

closed 2020-11-17 by yann44