righthalfplane/SdrGlut #14: Could not compile on Ubuntu 20.04

yeckel, 2020-11-24:
I've installed the packages from Readme and getting:

```s -I./Classes -I./SdrGlut -I./FMRadio  -I./Radio   -c -o Radio/Radio.o Radio/Radio.cpp
Radio/Radio.cpp: In function ‘int doWindow(double*, double*, long int, int)’:
Radio/Radio.cpp:2522:22: error: ‘liquid_hann’ was not declared in this scope
 2522 |                 w[i]=liquid_hann(i, (int)length);
      |                      ^~~~~~~~~~~
Radio/Radio.cpp:2535:22: error: ‘liquid_hamming’ was not declared in this scope; did you mean ‘liquid_gammaf’?
 2535 |                 w[i]=liquid_hamming(i, (int)length);
      |                      ^~~~~~~~~~~~~~
      |                      liquid_gammaf
Radio/Radio.cpp:2546:22: error: ‘liquid_flattop’ was not declared in this scope; did you mean ‘liquid_factor’?
 2546 |                 w[i]=liquid_flattop(i, (int)length);
      |                      ^~~~~~~~~~~~~~
      |                      liquid_factor
Radio/Radio.cpp:2558:22: error: ‘liquid_blackmanharris’ was not declared in this scope; did you mean ‘blackmanharris’?
 2558 |                 w[i]=liquid_blackmanharris(i, (int)length);
      |                      ^~~~~~~~~~~~~~~~~~~~~
      |                      blackmanharris
Radio/Radio.cpp:2569:22: error: ‘liquid_blackmanharris7’ was not declared in this scope; did you mean ‘blackmanharris7’?
 2569 |                 w[i]=liquid_blackmanharris7(i, (int)length);
      |                      ^~~~~~~~~~~~~~~~~~~~~~
      |                      blackmanharris7
make: *** [<builtin>: Radio/Radio.o] Error 1
```
I've deleted the  ```#define WINDOWS_LONG_NAMES``` from MainSource/firstFile.h and now it compiles. Installed libliquid-dev is in version 1.3.2-2

righthalfplane, 2020-11-24:
I have tried to fix that problem, but across the systems libliquid is inconsistent in its numbering of versions. That gives me a thought - I look in libliquid to see how he does it - unfortunately much of libliquid is created on the fly by macro expansion - so I may not be able to find it.

yeckel, 2020-11-25:
Then maybe a hint into readme?

righthalfplane, 2020-11-25:
Yes, I will have to redo the readme. libliquid has the same version number on different version.

closed 2020-11-27 by righthalfplane