righthalfplane/SdrGlut #16: Distorted audio

srs4511351, 2020-11-28:
I installed SdrGlut-1.19.zip on another Raspberry Pi 4 / SDRplay RSP1A system than I used in my previous post: "Starts to run, but hangs when starting the SDR".
This installation reports version SdrGlut-721. It recognizes my SDR without having to use Soapy Remote. SdrGlut runs and receives signals.
The sound is extremely garbled. In addition, when SdrGlut is running while I play a youtube video, it gables that sound also. If I quit SdrGlut, the sound in the video immediately clears up. 
The test playsaw command specified in the RtAudio youtube video: ./playsaw 1 40000 does not work, giving  the error message RtApiPulse::probeDeviceOpen: unsupported sample rate. ./playsaw 1 44100 works.

What is wrong here? Is there a way to configure the audio/sample rate in SdrGlut?

righthalfplane, 2020-11-29:
I just put out a new version (755) that fixes problems with SoapyRemote.

git clone https://github.com/righthalfplane/SdrGlut.git

I compiles for me on a Raspberry pi 4 with no changes to the source code.

The garbled sound is caused by pulseaudio. To cure the problem remove pulseaudio ie -

sudo apt-get remove pulse audio

then restart your system.

srs4511351, 2020-11-29:
Thanks for your quick reply. Removing pulseaudio did fix the audio problem.
Now I have to deal with my other applications, that either require pulseaudio or are compiled to use it.

I found that compile will fail if I have libliquid-dev v1.3.2 from https://github.com/jgaeddert/liquid-dsp
It does compile with version 1.3.1-1 installed with the system packager. I prefer to use the latest, though for CubicSDR.

The new version (755) works better than ever. It recognizes my RSP1A without having to use Soapy Remote.

I noticed that I am receiving what I will call bleedthrough from the sideband opposite from what I am using when there is a signal there. Some of my SDR applications have this problem and others handle it quite well.

righthalfplane, 2020-11-29:
Pulseaudio is strange - on the PI 4 it has to be off - on Ubuntu 18.04 everything works fine with it - on Ubuntu 20.04 things mostly work with it on. There may be someway to get it to work on the PI 4 ?

SdrGlut will build with any version of libliquid - you just have to fiddle with the LIQUID_VERSION_4 and WINDOWS_LONG_NAMES flags in the file MainSource/firstFile.h. 

The USB and LSB bleedthrough was fixed in the later versions libliquid - I complained and sent him an example and he fixed it - I get only a tiny  bleedthrough on transmit or receive with version 1.3.2 and SdrGlut on MacOS High Sierra.

srs4511351, 2020-11-29:
I may have to figure out pulseaudio. wsjt-x and QRadiolink require it.
I have thought of some possibilities.

Could we make a script to start SdrGlut that stops pulseaudio and starts it
when SdrGlut quits?
I wouldn't be able to use wasn't-x with SdrGllut.

pulseaudio-alsa, not available in the Raspberry Pi packager.
https://wiki.parabola.nu/PulseAudio
This package contains the necessary /etc/asound.conf for configuring ALSA
to use PulseAudio.
pulseaudio-alsa, provides compatibility layer between ALSA applications and
PulseAudio.
pulseaudio-alsa is part of alsa-plugins or a split package

What I found was an archive that contains system files to be extracted to
the system folders. I don't know how to properly install this sort of
thing.

I did try following the instructions on your GitHub web site for
MainSource/firstFile.h.
It didn't help on my first try. I'll need to try again.

----Steve

righthalfplane, 2020-11-30:
I was able to leave pulseaudio running by installing and using alsoft-conf -

sudo apt-get install alsoft-conf
sudo alsoft-conf

and disabling all of the backend except for "ALSA" - I get no more distortion.

srs4511351, 2020-12-01:
Thanks! This helps.
I tried it, but only disabled the pulse backend.
SdrGlut works and the applications that use pulse work.
I was able to use SdrGlut to decode FT8 with wsjt-x

----Steve

righthalfplane, 2020-12-02:
Great ! You should make a video showing how to do it .

srs4511351, 2020-12-02:
I am working on a document with instructions on how to do this on the
Raspberry Pi.
It includes several other SDR applications also.
I can post a copy.

----Steve

righthalfplane, 2020-12-03:
Let me know when and where - so I can take a look at it.

srs4511351, 2020-12-05:
I have the procedure to get wsjt-x working with SdrGlut.
Depending on how your system is configured, you may only need to tune
SdrGlut to the digital signals and select the right audio input device in
WSJT-X.
Here is the procedure that I used. Alternately, if you want the audio to be
routed directly to WSJT-X, I can configure it with a virtual audio device.
Note that CubicSDR is nice because it can have many demodulators or radios
at the same time. I often use that to map FT8 stations while listening to
SSB QSOs on the same band.

Go here:
http://web.eecs.umich.edu/~srs/tech.html
Look for the SdrGlut section on the page:
Part 3 - Setting up WSJT-X to work with your SDR program
<http://web.eecs.umich.edu/~srs/digital_modes.html>

I have instruction on building a working system for SDRplay from scratch to
installing various SDR programs, etc.

----Steve

righthalfplane, 2020-12-05:
Thats great I will take a look at it !

SdrGlut has a frequency scanning mode that you may find useful. I can build  lists of frequencies that can be read into SdrGlut. I have lists for the analog emergency services and for the digital emergency services and many other things. With the p25 and other digital signals, I can pipe the signals to szechyjs-dsd or LouisErigHere-dsd for decoding. Here is a video showing some examples -

The Improved Search and Scan Features of SdrGlut
https://youtu.be/4oTv2DinCRk

srs4511351, 2021-02-09:
I have a new installation on ubuntu-20.04.1 and am again having problems
with distorted audio with SdrGlut.
You gave me a solution by installing and using alsoft-conf and disabling
all of the backend except for "ALSA".

However, alsoft-conf is deprecated and is no longer available.

Do you know what this utility actually does? Maybe I can change
the configuration manually if I know what that is.
Is there another utility that can disable all of the backends except for
"ALSA"?

It's strange, though, my later installations on Raspberry Pi OS did not
have the distortion issue at all.
It looks like SdrGlut is using Pulse when it is distorted.

----Steve

righthalfplane, 2021-02-19:
alsoft-conf just sets things in the files, but I am not sure witch ones. I started to track it down, but the problem went away after a while on Ubuntu 20.04 - so I stopped looking.

closed 2021-02-23 by righthalfplane

srs4511351, 2021-02-23:
Thanks for the reference to the article. None of the suggestions in the article fixed garbled audio for me, but it helped a lot.
What confused me and possibly some of the contributors is that changing period_size to anything legal and saving the file fixed it until after a reboot. The problem always came back.
I did find 2 things that fixed it permanently.
1. Your original suggestion of running alsoft-conf and disabling all of the backends except for "ALSA"...
I found what it was doing in etc/openal/alsoft.conf. Changing the drivers line to specify the desired backends:
drivers = alsa,port,oss
This fixed it even after a reboot.
2. The mention of "the garbled output uses 3 buffer periods, whereas the ones that are okay use 4" by kcat led me to specify periods = 4 in etc/openal/alsoft.conf.
That also fixed the garbled audio after a reboot, even with the other items set to default.

----Steve