righthalfplane/SdrGlut #17: Can't compile on Linux Mint 19.3

janoleholm, 2020-12-10:
Want to try your sdr with my LimeSDR mini, but get this error:

Radio/RaceFastsdr.cpp:1199:50: error: invalid conversion from ‘int’ to ‘liquid_ampmodem_type’ [-fpermissive]
     f->demodAM = ampmodem_create(0.5, mode, iflag);
                                                  ^
Radio/RaceFastsdr.cpp:1199:50: error: too few arguments to function ‘ampmodem_s* ampmodem_create(float, float, liquid_ampmodem_type, int)’

righthalfplane, 2020-12-10:
In the file MainSource/firstFile.h comment out the "#if LIQUID_VERSION_NUMBER >= 1003001 || __arm__" like -

//#if LIQUID_VERSION_NUMBER >= 1003001 || __arm__
#define LIQUID_VERSION_4 1
//#endif

janoleholm, 2020-12-10:
Thanks very much.

Now I compile without errors :-)

closed 2020-12-10 by janoleholm