righthalfplane/SdrGlut #18: Compile and install SdrGlut in User Space

janoleholm, 2020-12-27:
First thanks for your efficient and good SDR-program. I use it with LimeSDR mini and USB3, and I want to use it on a Linux Mint laptop (ubuntu 18.04), where SoapySDR are installed in User Space (/home/jhj/projects/sdr_install). I managed to compile and install RtAudio in userspace with:

> cmake -DCMAKE_INSTALL_PREFIX=$LIME_INSTALL -DCMAKE_PREFIX_PATH=$LIME_INSTALL ..

Tests/playsaw 1 48000 of RtAudio is working.
I am newbie on make and don't know how to fix your fireball in makefileUbunta with parameter -lrtaudio -lSoapySDR.

Another question is if you would create a user forum in Groups.io for your SdrGlut ?

73 OZ1OY

janoleholm, 2020-12-27:
I made it with this -L options:
-L/usr/local -L/usr/X11/lib -L/home/jhj/projects/sdr_install/lib 

But still a user forum would be good. I have more questions :-)

righthalfplane, 2020-12-27:
I like to keep the questions here. I look at many other locations, but rarely get any comments.

https://discourse.myriadrf.org
https://groups.google.com/g/gqrx
https://groups.google.com/g/pothos-users
https://www.raspberrypi.org/forums/viewforum.php?f=41&sid=0709b7b1d76c6f3127e2ba4708fdcf42
https://sdrplayusers.net/forums/
https://forums.radioreference.com/forums/software-defined-radio.193/

janoleholm, 2020-12-27:
Ok, you decide. I am not familiar with Github, and I see your are busy in many forum :-)

Unfortunately I got segmentation error, and I have SdrGlut running on a Raspberry Pi 4 and another Ubuntu (now 20.10). So I could stick with that.

```
[INFO] Make connection: 'LimeSDR Mini [USB 3.0] 1D42400478A0C2'
[INFO] Reference clock 40.00 MHz
[INFO] Device name: LimeSDR-Mini
[INFO] Reference: 40 MHz
[INFO] LMS7002M register cache: Disabled
device 0x564363244320
receive channels 1 transmit channels 1
Hardware boardSerialNumber: 0x1d42400478a0c2
Hardware firmwareVersion: 6
Hardware gatewareVersion: 1.30
Hardware hardwareVersion: 2
Hardware protocolVersion: 1
Segmentation fault (core dumped)
```

righthalfplane, 2020-12-28:
Which system are you getting the error on (ubuntu 18.04)? 
I am running a LimeSDR Mini on ubuntu 18.04 with no problems.
Did you follow the "Installation steps on Ubuntu 18.04" that I suggested in the ReadME ?

janoleholm, 2020-12-28:
`ldd sdrglut.x
	linux-vdso.so.1 (0x00007ffe6e6ab000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007fb3a49aa000)
	libglut.so.3 => /usr/lib/x86_64-linux-gnu/libglut.so.3 (0x00007fb3a4761000)
	libGL.so.1 => /usr/lib/x86_64-linux-gnu/libGL.so.1 (0x00007fb3a44d5000)
	libGLU.so.1 => /usr/lib/x86_64-linux-gnu/libGLU.so.1 (0x00007fb3a4266000)
	libopenal.so.1 => /usr/lib/x86_64-linux-gnu/libopenal.so.1 (0x00007fb3a3fc7000)
	librtaudio.so.6 => /home/jhj/projects/sdr_install/lib/librtaudio.so.6 (0x00007fb3a3d9d000)
	libSoapySDR.so.0.8 => /home/jhj/projects/sdr_install/lib/libSoapySDR.so.0.8 (0x00007fb3a3b2b000)
	libliquid.so.1d => /usr/lib/x86_64-linux-gnu/libliquid.so.1d (0x00007fb3a3870000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007fb3a34e7000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007fb3a3149000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007fb3a2f31000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fb3a2b40000)
	/lib64/ld-linux-x86-64.so.2 (0x00007fb3a531e000)
	libX11.so.6 => /usr/lib/x86_64-linux-gnu/libX11.so.6 (0x00007fb3a2808000)
	libXi.so.6 => /usr/lib/x86_64-linux-gnu/libXi.so.6 (0x00007fb3a25f8000)
	libXxf86vm.so.1 => /usr/lib/x86_64-linux-gnu/libXxf86vm.so.1 (0x00007fb3a23f2000)
	libGLX.so.0 => /usr/lib/x86_64-linux-gnu/libGLX.so.0 (0x00007fb3a21c1000)
	libGLdispatch.so.0 => /usr/lib/x86_64-linux-gnu/libGLdispatch.so.0 (0x00007fb3a1f0b000)
	libsndio.so.6.1 => /usr/lib/x86_64-linux-gnu/libsndio.so.6.1 (0x00007fb3a1cfb000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007fb3a1af3000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007fb3a18ef000)
	libasound.so.2 => /usr/lib/x86_64-linux-gnu/libasound.so.2 (0x00007fb3a15e8000)
	libpulse.so.0 => /usr/lib/x86_64-linux-gnu/libpulse.so.0 (0x00007fb3a1398000)
	libpulse-simple.so.0 => /usr/lib/x86_64-linux-gnu/libpulse-simple.so.0 (0x00007fb3a1193000)
	libfftw3f.so.3 => /usr/lib/x86_64-linux-gnu/libfftw3f.so.3 (0x00007fb3a0d86000)
	libxcb.so.1 => /usr/lib/x86_64-linux-gnu/libxcb.so.1 (0x00007fb3a0b5e000)
	libXext.so.6 => /usr/lib/x86_64-linux-gnu/libXext.so.6 (0x00007fb3a094c000)
	libbsd.so.0 => /lib/x86_64-linux-gnu/libbsd.so.0 (0x00007fb3a0737000)
	libpulsecommon-11.1.so => /usr/lib/x86_64-linux-gnu/pulseaudio/libpulsecommon-11.1.so (0x00007fb3a04b8000)
	libdbus-1.so.3 => /lib/x86_64-linux-gnu/libdbus-1.so.3 (0x00007fb3a026b000)
	libXau.so.6 => /usr/lib/x86_64-linux-gnu/libXau.so.6 (0x00007fb3a0067000)
	libXdmcp.so.6 => /usr/lib/x86_64-linux-gnu/libXdmcp.so.6 (0x00007fb39fe61000)
	libsystemd.so.0 => /lib/x86_64-linux-gnu/libsystemd.so.0 (0x00007fb39fbdd000)
	libwrap.so.0 => /lib/x86_64-linux-gnu/libwrap.so.0 (0x00007fb39f9d3000)
	libsndfile.so.1 => /usr/lib/x86_64-linux-gnu/libsndfile.so.1 (0x00007fb39f75a000)
	libasyncns.so.0 => /usr/lib/x86_64-linux-gnu/libasyncns.so.0 (0x00007fb39f554000)
	libapparmor.so.1 => /lib/x86_64-linux-gnu/libapparmor.so.1 (0x00007fb39f344000)
	liblzma.so.5 => /lib/x86_64-linux-gnu/liblzma.so.5 (0x00007fb39f11e000)
	liblz4.so.1 => /usr/lib/x86_64-linux-gnu/liblz4.so.1 (0x00007fb39ef02000)
	libgcrypt.so.20 => /lib/x86_64-linux-gnu/libgcrypt.so.20 (0x00007fb39ebe6000)
	libnsl.so.1 => /lib/x86_64-linux-gnu/libnsl.so.1 (0x00007fb39e9cc000)
	libFLAC.so.8 => /usr/lib/x86_64-linux-gnu/libFLAC.so.8 (0x00007fb39e755000)
	libogg.so.0 => /usr/lib/x86_64-linux-gnu/libogg.so.0 (0x00007fb39e54c000)
	libvorbis.so.0 => /usr/lib/x86_64-linux-gnu/libvorbis.so.0 (0x00007fb39e321000)
	libvorbisenc.so.2 => /usr/lib/x86_64-linux-gnu/libvorbisenc.so.2 (0x00007fb39e078000)
	libresolv.so.2 => /lib/x86_64-linux-gnu/libresolv.so.2 (0x00007fb39de5e000)
	libgpg-error.so.0 => /lib/x86_64-linux-gnu/libgpg-error.so.0 (0x00007fb39dc49000)
`
Laptop PC is:
`inxi -S
System:
  Host: Aspire-one Kernel: 5.4.0-58-generic x86_64 bits: 64 
  Desktop: Cinnamon 4.4.8 Distro: Linux Mint 19.3 Tricia 
`
which should be a ubuntu 18.04 variant.
I have installed Gqrx on a full Soapy source stack following this thread:
[http://blog.reds.ch/?p=43](url)
and Gqrx work on a Funcube, an RTL-dongle and my LimeSDR mini.
I know your installation procedure from installing SdrGlut on a Ubuntu 20.10 and a Raspberry Pi 4, but on my Linux Mint only installed packaged from your procedure, that I didn't have in my source stack.

righthalfplane, 2020-12-28:
Because you put your library at "/home/jhj/projects/sdr_install", you need to set the library path -

With tcsh -
setenv LD_LIBRARY_PATH /home/jhj/projects/sdr_install/lib

With bash
export LD_LIBRARY_PATH =/home/jhj/projects/sdr_install/lib

librtaudio was not in the "ldd sdrglut.x" list.

janoleholm, 2020-12-28:
This is the end of my .bashrc:
`export LIME_INSTALL=~/projects/sdr_install
export LIME_SRC=~/projects/sdr_src
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:$LIME_INSTALL/lib
export PATH=$PATH:$LIME_INSTALL/bin`

And I think librtaudio is in my file:
`ldd sdrglut.x |grep rtaudio
	librtaudio.so.6 => /home/jhj/projects/sdr_install/lib/librtaudio.so.6 (0x00007f057f32f000)
`
Thanks for taking your time trying to help me :-)

closed 2020-12-29 by righthalfplane