righthalfplane/SdrGlut #19: Command line control of SdrGlut

rrobinett, 2021-01-15:
Thanks, I am very pleased to find this excellent package
However for my wsprdaemon bash application (http://wsprdaemon.org/) I need to record a continuous series of 12000 sps raw audio files.
Could SdrGlut be configured or easily modified for my application?
The rx_tools/rx_fm package which claims to do that has serious audio and frequency tuning problems which I have been unable to fix.  SdrGlut suffers from neither of those problems and would be an ideal addition to Wsprdaemon.
Thanks
Rob

righthalfplane, 2021-01-15:
It would likely easy to modify SdrGlut to do what you need, but sdrTest may already have want you need. Here is a video showing some of its capabilities.

https://youtu.be/hxxhsxCiPlY

Dale.

rrobinett, 2021-01-15:
Hi Dale,

Thanks for your quick reply.  Your sdrTest appears to do exactly what I need, but:

1) I get a 'out of memory' error when I run sdrGlut on a rebooted 4 GB Pi 4. 
2) I don't see a makefile or other mechanism to create sdrTest

Thanks,

Rob

=========================================================
> pi@Wsprdaemon-Pi:~/SdrGlut $ ./sdrglut.x 
LIQUID_VERSION_NUMBER 1003000
Lib Version: v0.6.1-4+b4
API Version: v0.6.0
ABI Version: v0.6
Install root: /usr
Search path: /usr/lib/arm-linux-gnueabihf/SoapySDR/modules0.6
Search path: /usr/local/lib/arm-linux-gnueabihf/SoapySDR/modules0.6
Search path: /usr/local/lib/SoapySDR/modules0.6
Module found: /usr/lib/arm-linux-gnueabihf/SoapySDR/modules0.6/libHackRFSupport.so
Module found: /usr/lib/arm-linux-gnueabihf/SoapySDR/modules0.6/libLMS7Support.so
Module found: /usr/lib/arm-linux-gnueabihf/SoapySDR/modules0.6/libRedPitaya.so
Module found: /usr/lib/arm-linux-gnueabihf/SoapySDR/modules0.6/libairspySupport.so
Module found: /usr/lib/arm-linux-gnueabihf/SoapySDR/modules0.6/libaudioSupport.so
Module found: /usr/lib/arm-linux-gnueabihf/SoapySDR/modules0.6/libbladeRFSupport.so
Module found: /usr/lib/arm-linux-gnueabihf/SoapySDR/modules0.6/libosmosdrSupport.so
Module found: /usr/lib/arm-linux-gnueabihf/SoapySDR/modules0.6/libremoteSupport.so
Module found: /usr/lib/arm-linux-gnueabihf/SoapySDR/modules0.6/librtlsdrSupport.so
Module found: /usr/lib/arm-linux-gnueabihf/SoapySDR/modules0.6/libuhdSupport.so
[ERROR] SoapySSDPEndpoint::sendTo(udp://[ff02::c]:1900) = -1
  sendto(udp://[ff02::c]:1900) [99: Cannot assign requested address]
Found Rafael Micro R820T tuner
[INFO] [UHD] linux; GNU C++ version 8.2.0; Boost_106700; UHD_3.13.1.0-3
default_input=False
default_output=False
device_id=2
driver=audio
label=default
available=Yes
driver=rtlsdr
label=Generic RTL2832U OEM :: 00000001
manufacturer=Realtek
product=RTL2838UHIDIR
rtl=0
serial=00000001
tuner=Rafael Micro R820T
Found Rafael Micro R820T tuner
receive channels 1 transmit channels 0
available=Yes
driver=rtlsdr
label=Generic RTL2832U OEM :: 00000001
manufacturer=Realtek
product=RTL2838UHIDIR
rtl=0
serial=00000001
tuner=Rafael Micro R820T
Found Rafael Micro R820T tuner
device 0x1985ac8
receive channels 1 transmit channels 0
Hardware origin: https://github.com/pothosware/SoapyRTLSDR
Hardware rtl: 0
cMalloc Out of Memory Requested (-1440541728) Total Used (1428527576) tag (8891)
cMalloc Out of Memory Requested (-1440541728) Total Used (1428527576) tag (8891)
cMalloc Out of Memory Requested (-1440541728) Total Used (1428527576) tag (8892)
cMalloc Out of Memory Requested (-1440541728) Total Used (1428527576) tag (8892)
Segmentation fault
pi@Wsprdaemon-Pi:~/SdrGlut $

righthalfplane, 2021-01-15:
The build instructions are at the top of sdrTest.cpp -

g++ -O2 -std=c++11 -Wno-deprecated -o sdrTest sdrTest.cpp mThread.cpp cMalloc.c -lrtaudio -lSoapySDR -lliquid -lpthread -lopenal

Your RTL stick returned bogus information for the listGains - try plugging and unplugging the device and If that does not shut everything down and bring it back up. Let me know what happens.

rrobinett, 2021-01-15:
1)  Perhaps the 'out of memory' error derives from my previous installation
of the Pothosware SoapySDR packages. I am preparing a clean Pi OS
installation as a work around.
2) I get a fatal compilation error:

Thanks for your help,

Rob AI6VN

============

pi@Wsprdaemon-Pi:~/SdrGlut/sdrTest $ g++ -O2 -std=c++11 -Wno-deprecated -o
sdrTest sdrTest.cpp mThread.cpp cMalloc.c -lrtaudio -lSoapySDR -lliquid
-lpthread -lopenal
sdrTest.cpp: In function ‘int setFilters(playData*, Filters*)’:
sdrTest.cpp:1270:45: error: invalid conversion from ‘int’ to
‘liquid_ampmodem_type’ [-fpermissive]
     f->demodAM = ampmodem_create(0.5, mode, iflag);
                                             ^~~~~
sdrTest.cpp:1270:50: error: too few arguments to function ‘ampmodem_s*
ampmodem_create(float, float, liquid_ampmodem_type, int)’
     f->demodAM = ampmodem_create(0.5, mode, iflag);
                                                  ^
In file included from sdrTest.cpp:9:
/usr/include/liquid/liquid.h:5906:10: note: declared here
 ampmodem ampmodem_create(float _m,
          ^~~~~~~~~~~~~~~
pi@Wsprdaemon-Pi:~/SdrGlut/sdrTest $

righthalfplane, 2021-01-15:
Put in a 

#define LIQUID_VERSION_4 1

at the top of sdrTest.cpp.

This is a problem with the liquid-dsp libraries not being correctly identified.

rrobinett, 2021-01-15:
Great.  That fixed the sdrTest compile problem.
Now, on the Pi where SdrGlut crashes when accessing the RTL-SDR, I get a
seg fault when I run sdrTest after it tries to list the RTL info.
I will restore the Pi and do a clean install of SdrGlut.  I'm sure that
will fix it.
Thanks for your help

Rob

rrobinett, 2021-01-15:
Unfortunately on a Pi with a fresh Raspberry OS 2021-01-11 image,  my compile of SdrGlub fails:

pi@WsprdaemonPi:~/SdrGlut $ make -f makefileRaspbian 
cc -O3 -DNO_FORTRAN -pthread -I/usr/local/hdf-64/include -I/usr/X11R6/include  -I./DrawFindPointHistory -I./MainSource -I./KdTree-routines -I./glui-routines/include -I./CExpress -I./FMRadio -I./Radio   -c -o MainSource/Getpalette.o MainSource/Getpalette.c
g++ -O3 -std=gnu++11 -DNO_FORTRAN  -I./Digital -pthread -I./glui-routines/include -I/usr/local/hdf-64/include -I/usr/X11R6/include  -I./DrawFindPointHistory  -I ./SdrFile -I./MainSource -I./KdTree-routines -I./glui-routines/include -I./CExpress -I./Classes -I./SdrGlut -I./FMRadio  -I./Radio   -c -o MainSource/mThread.o MainSource/mThread.cpp
cc -O3 -DNO_FORTRAN -pthread -I/usr/local/hdf-64/include -I/usr/X11R6/include  -I./DrawFindPointHistory -I./MainSource -I./KdTree-routines -I./glui-routines/include -I./CExpress -I./FMRadio -I./Radio   -c -o MainSource/Scene.o MainSource/Scene.c
In file included from MainSource/Scene.c:10:
MainSource/Scene.h:20:10: fatal error: GL/glut.h: No such file or directory
 #include <GL/glut.h> /* GLUT OpenGL includes */
          ^~~~~~~~~~~
compilation terminated.
make: *** [<builtin>: MainSource/Scene.o] Error 1
pi@WsprdaemonPi:~/SdrGlut $

rrobinett, 2021-01-15:
That compile problem was fixed with:
          sudo apt install freeglut3-dev

The compilation proceeds much farther until it encounters this error, which I expect requires another package to be loaded:

++ -O3 -std=gnu++11 -DNO_FORTRAN  -I./Digital -pthread -I./glui-routines/include -I/usr/local/hdf-64/include -I/usr/X11R6/include  -I./DrawFindPointHistory  -I ./SdrFile -I./MainSource -I./KdTree-routines -I./glui-routines/include -I./CExpress -I./Classes -I./SdrGlut -I./FMRadio  -I./Radio   -c -o MainSource/QuitDialog.o MainSource/QuitDialog.cpp
In file included from MainSource/QuitDialog.cpp:18:
MainSource/audiolib.h:15:10: fatal error: AL/alut.h: No such file or directory
 #include <AL/alut.h>
          ^~~~~~~~~~~
compilation terminated.
make: *** [<builtin>: MainSource/QuitDialog.o] Error 1
pi@WsprdaemonPi:~/SdrGlut $

rrobinett, 2021-01-15:
I'm sorry to have polluted the forum.  In documenting the installation steps I failed to execute the 'apt install's you document are needed.
After a clean install of the Raspberry OS I now have both SdrGlut and sdrTest running on my Pi4.

Can you give me any hints on how to change the output audio sample rate down to 12000 sps and write the audio samples to a raw file rather than to the audio output port?

Thanks for your help

Rob

righthalfplane, 2021-01-16:
If you refresh your copy of the SdrGlut files, there is a new version of sdrTest. It will write 12000 sps to a files with -

./sdrTest -fc 102.0e6 -f 102.1e6 -fm -gain 1 -faudio 12000 -file test.raw

I tested the file with Audacity and it played correctly. This only works for writing files. Playing the audio works as before.

rrobinett, 2021-01-16:
Thanks for your efforts.  Unfortunately I can't compile the new sdrTest on my Pi:

pi@WsprdaemonPi:~/SdrGlut/sdrTest $ g++ -O2 -std=c++11 -Wno-deprecated -o sdrTest sdrTest.cpp mThread.cpp cMalloc.c -lrtaudio -lSoapySDR -lliquid -lpthread -framework OpenAL
g++: error: OpenAL: No such file or directory
g++: error: unrecognized command line option ‘-framework’
pi@WsprdaemonPi:~/SdrGlut/sdrTest $ g++ -O2 -std=c++11 -Wno-deprecated -o sdrTest sdrTest.cpp mThread.cpp cMalloc.c -lrtaudio -lSoapySDR -lliquid -lpthread -lopenal
sdrTest.cpp: In function ‘int setFilters(playData*, Filters*)’:
sdrTest.cpp:1279:45: error: invalid conversion from ‘int’ to ‘liquid_ampmodem_type’ [-fpermissive]
     f->demodAM = ampmodem_create(0.5, mode, iflag);
                                             ^~~~~
sdrTest.cpp:1279:50: error: too few arguments to function ‘ampmodem_s* ampmodem_create(float, float, liquid_ampmodem_type, int)’
     f->demodAM = ampmodem_create(0.5, mode, iflag);
                                                  ^
In file included from sdrTest.cpp:9:
/usr/include/liquid/liquid.h:5906:10: note: declared here
 ampmodem ampmodem_create(float _m,
          ^~~~~~~~~~~~~~~
pi@WsprdaemonPi:~/SdrGlut/sdrTest $

rrobinett, 2021-01-16:
Also, could I make one final feature request?

The WSPR packets are 2, 5, 15, or 30 minutes in length and always start at second 0 of a minute, so to decode them I need to generate a series of one minute long audio files which start at second zero of each minute.  Also, in order to use standard RTL-SDRs for VHF/UHF WSPR reception, I tune in USB mode to the pilot carrier of local ATSC TV stations and measure the audio frequency.  From the frequency of the resulting audio tone I can calibrate the RTL-SDR's oscillator error and use that when tuning it to the VHF/UHF WSPR bands.

So  would it be possible for you to add a raw file output mode which creates a series of such one minute long files?  I can then 'cat' them into the 2/5/15/30 minute wav files which are processed by the 'wsprd' utility included in the WSJT-x package.  In WD I will spawn a 'sdrTest job which will run forever, while the parent job will monitor for the time-stamped raw audio files and assemble them for decoding by 'wsprd'.
For calibration I could 'timeout 1 sdrTest...', but it would be much cleaner if there were a '--timeout N' flag, where N is in msecs or secs.

These features will enable Wsprdaemon to use all the SDRs which are supported by Soapy drivers and thus extend WD support from the current 0-30Mhz bands up through VHF/UHF WSPR bands.  There are dozens of Hams around the world who have been waiting for me to add support for non-KiwiSDR receivers, and with sdrTest we will be able to make some very interesting studies comparing the WSPR reception performance of a wide range of SDRs.  In addition, offering a low cost VHF/UHF WSPR reception system using a Pi + 4 (or more) RTL-SDRs (and other SDR) should encourage many more VHF/UHF WSPR deployments and lead to new insights into propagation on those bands.

If it would be helpful to you, I can create an account on my development Pi and give you remote access.

Thanks,

Rob

righthalfplane, 2021-01-17:
The new README.txt should answer your compile questions - let me know if it doesn't.

Currently it takes about 1/10 second to fill the I/O buffer and 4/10 of a second for the agc to stabilize - so it is about 1/2 second to get things going and when you stop - it is aways + or - 1/10 of a second. If you need anything better than that it will require another level of buffering.

rrobinett, 2021-01-17:
Sorry, I forgot to uncomment that line after the 'git pull'.  It now compiles and runs fine.

The startup timing is not critical at all.
In normal operation I would like to start sdrTest and leave it running for hours (depending upon the stability of the oscillator) on the same frequency outputting a series of one minute  long audio files.
Occasionally I would interrupt that sdrTest job and tune to a frequency with a known good pilot tone and record a few seconds of it to determine if the oscillator drifted.  After calculating the oscillator error I would return to recording another hour.

righthalfplane, 2021-01-17:
I added the -timeout in seconds and the -dumpbyminute options to sdrTest. The timeout defaults to infinity.

Record one minute of audio to the file test.raw.
./sdrTest -fc 102.0e6 -f 102.1e6 -fm -gain 1 -timeout 60 -faudio 12000 -file test.raw

Write the sound data into 1 minute files starting at the next minute and continue for the next 10 minutes.
./sdrTest -fc 102.0e6 -f 102.1e6 -fm -gain 1 -timeout 601 -dumpbyminute -faudio 12000

The files are of the form minute-00000001.raw where the number is the number minutes from the start minute. This writes one addition file that is one second to make sure you get the first 10 minutes.

Write the minute files until it gets a control c.
./sdrTest -fc 102.0e6 -f 102.1e6 -fm -gain 1 -dumpbyminute -faudio 12000

rrobinett, 2021-01-17:
Excellent!  Thank you so much.
I am using your latest code to verify the oscillator accuracy by tuning 1000 hz below the pilot carrier and then using 'sox -n stat -freq' to measure the resulting audio frequency.  Knowing that error, I can then tune to within a few hz of the 54/144/432 Mhz WSPR bands, which are only 200 Hz wide:

pi@WsprdaemonPi:~/soapy-test $ ./sdr-test.sh
Soapy device #1 tuned to the pilot signal of TV RF channel # 7 and got audio at  1076 Hz, so audio error is  76 Hz at RF frequency 174308441 =>  2.29 ppm oscillator error.  Mean level 0.253121
Soapy device #1 tuned to the pilot signal of TV RF channel #13 and got audio at  1083 Hz, so audio error is  83 Hz at RF frequency 210308441 =>  2.53 ppm oscillator error.  Mean level 0.246205
Soapy device #1 tuned to the pilot signal of TV RF channel #19 and got audio at  1486 Hz, so audio error is 486 Hz at RF frequency 500308441 =>  1.02 ppm oscillator error.  Mean level 0.179980
Soapy device #1 tuned to the pilot signal of TV RF channel #27 and got audio at  1461 Hz, so audio error is 461 Hz at RF frequency 548308441 =>  1.18 ppm oscillator error.  Mean level 0.236698
Soapy device #1 tuned to the pilot signal of TV RF channel #29 and got audio at  1340 Hz, so audio error is 340 Hz at RF frequency 560308441 =>  1.64 ppm oscillator error.  Mean level 0.251788
Soapy device #1 tuned to the pilot signal of TV RF channel #30 and got audio at  1632 Hz, so audio error is 632 Hz at RF frequency 566308441 =>  0.89 ppm oscillator error.  Mean level 0.257407
Soapy device #1 tuned to the pilot signal of TV RF channel #33 and got audio at  1495 Hz, so audio error is 495 Hz at RF frequency 584308441 =>  1.18 ppm oscillator error.  Mean level 0.233985
Soapy device #1 tuned to the pilot signal of TV RF channel #38 and got audio at  1048 Hz, so audio error is  48 Hz at RF frequency 614308441 => 12.79 ppm oscillator error.  Mean level 0.144804
Soapy device #1 tuned to the pilot signal of TV RF channel #39 and got audio at  1059 Hz, so audio error is  59 Hz at RF frequency 620308441 => 10.51 ppm oscillator error.  Mean level 0.139501
Soapy device #1 tuned to the pilot signal of TV RF channel #40 and got audio at  1153 Hz, so audio error is 153 Hz at RF frequency 626308441 =>  4.09 ppm oscillator error.  Mean level 0.136750
Soapy device #1 tuned to the pilot signal of TV RF channel #41 and got audio at   981 Hz, so audio error is -19 Hz at RF frequency 632308441 => -33.27 ppm oscillator error.  Mean level 0.132242
Soapy device #1 tuned to the pilot signal of TV RF channel #43 and got audio at  1127 Hz, so audio error is 127 Hz at RF frequency 644308441 =>  5.07 ppm oscillator error.  Mean level 0.127359
Soapy device #1 tuned to the pilot signal of TV RF channel #44 and got audio at  1124 Hz, so audio error is 124 Hz at RF frequency 650308441 =>  5.24 ppm oscillator error.  Mean level 0.131724
pi@WsprdaemonPi:~/soapy-test $

righthalfplane, 2021-01-18:
Are you getting a different error correction off of the same device for each of the TV stations ?

rrobinett, 2021-01-18:
I don't know the frequency accuracy of those pilot carriers, so I am testing with a RTL-SDR which I have modified to  be driven by my Bodnar GPSDO.
One Bodnar output is configured to output the 28.8 Mhz clock fed to the RTL-SDR clock input.
The second Bodnar output is set to output 288.0 Mhz.
The Bodnar frequencies are accurate to almost 1e-12 , and of course they are synchronized with each other.
However when I configure sdrTest to tune USB at 287.999Mhz, I get a  1049Hz tone, not the 1000 Hz I would expect.
I am now enhancing my bash script to tune +-500/200/100/50/20/10 Hz to see if the tone changes in reasonable sync with the RTL-SDR tuning frequency.
Since I haven't studied the RTL chip, I'm not sure what tuning accuracy I should expect.
Here is how i am measuring:

sdrTest -fc 287e6 -f 287999000 -usb -gain 1 -device 1 -faudio 12000 -timeout 2  -file sdraudio.raw 
cat sdraudio.raw | sox -r 12000 -t raw -e s -b 16 -c 1 - -n stat -freq > sox.txt 2>&1

rrobinett, 2021-01-19:
Hi Dale,

Tuning accuracy is the final challenge.  My tests show that on the same Pi and with the same GPSDO RTL-SDR, CubicSDR tunes within 40Hz at 288 Mhz, and small changes in tuning are accurately reflected in changes in the output audio frequency.  sdrTest is generally less accurate and the changes in audio error are larger (200+ Hz) and hard to predict when changing the tuning frequency.

The fact that changing the tuning frequency 2 or 5 Hz with Cubic results in a corresponding 2 or 5 Hz change in audio suggests that the RTL-STL is capable of accurate and repeatable tuning.  If so, then perhaps there are different rounding errors between Cubic and sdrTest in the calculations which transform the tuning frequency to the values programmed into the RTL's registers.

Here are the measurements results which led me to that theory. In all tests I use the first output of the GPSDO to drive the clock of the RTL-SDR and the second output of the GPSDO to generate a known accurate RF carrier frequency of 288 Mhz:

==================
Using sdrTest and the GPSDO RTL-SDR, I see frequency errors of +200Hz to -24 Hz as I tune from 1500 Hz below to 500 Hz below the 288 Mhz signal, and those errors are not monotonic.  When tuning up from 1500 Hz below, the audio differs by: +56, -24, +45.... +220Hz.  Since the WSPR band is only 200 Hz wide, these errors are two great for me to reliably tune to the 432.300MHz WSPR band.

pi@WsprdaemonPi:~/soapy-test $ ./sdr-test.sh
Mon 18 Jan 2021 04:02:04 PM PST: check_tuning_accuracy() test device #1 at 287999000, expecting 1000 Hz audio at that center tuning frequency
Mon 18 Jan 2021 04:02:11 PM PST: check_tuning_accuracy() device #1 tuned to frequency 287998500. Expected 1500 Hz audio, measured 1556 Hz audio, so tuning is off by   56 Hz =  0.19 ppm
Mon 18 Jan 2021 04:02:19 PM PST: check_tuning_accuracy() device #1 tuned to frequency 287998800. Expected 1200 Hz audio, measured 1176 Hz audio, so tuning is off by  -24 Hz = -0.08 ppm
Mon 18 Jan 2021 04:02:26 PM PST: check_tuning_accuracy() device #1 tuned to frequency 287998900. Expected 1100 Hz audio, measured 1145 Hz audio, so tuning is off by   45 Hz =  0.15 ppm
Mon 18 Jan 2021 04:02:33 PM PST: check_tuning_accuracy() device #1 tuned to frequency 287998950. Expected 1050 Hz audio, measured 1145 Hz audio, so tuning is off by   95 Hz =  0.32 ppm
Mon 18 Jan 2021 04:02:40 PM PST: check_tuning_accuracy() device #1 tuned to frequency 287998980. Expected 1020 Hz audio, measured 1145 Hz audio, so tuning is off by  125 Hz =  0.43 ppm
Mon 18 Jan 2021 04:02:47 PM PST: check_tuning_accuracy() device #1 tuned to frequency 287998990. Expected 1010 Hz audio, measured 1132 Hz audio, so tuning is off by  122 Hz =  0.42 ppm
Mon 18 Jan 2021 04:02:54 PM PST: check_tuning_accuracy() device #1 tuned to frequency 287999000. Expected 1000 Hz audio, measured 1145 Hz audio, so tuning is off by  145 Hz =  0.50 ppm
Mon 18 Jan 2021 04:03:02 PM PST: check_tuning_accuracy() device #1 tuned to frequency 287999010. Expected  990 Hz audio, measured 1145 Hz audio, so tuning is off by  155 Hz =  0.53 ppm
Mon 18 Jan 2021 04:03:09 PM PST: check_tuning_accuracy() device #1 tuned to frequency 287999020. Expected  980 Hz audio, measured 1145 Hz audio, so tuning is off by  165 Hz =  0.57 ppm
Mon 18 Jan 2021 04:03:16 PM PST: check_tuning_accuracy() device #1 tuned to frequency 287999050. Expected  950 Hz audio, measured 1145 Hz audio, so tuning is off by  195 Hz =  0.67 ppm
Mon 18 Jan 2021 04:03:23 PM PST: check_tuning_accuracy() device #1 tuned to frequency 287999100. Expected  900 Hz audio, measured 1145 Hz audio, so tuning is off by  245 Hz =  0.85 ppm
Mon 18 Jan 2021 04:03:31 PM PST: check_tuning_accuracy() device #1 tuned to frequency 287999200. Expected  800 Hz audio, measured  922 Hz audio, so tuning is off by  122 Hz =  0.42 ppm
Mon 18 Jan 2021 04:03:38 PM PST: check_tuning_accuracy() device #1 tuned to frequency 287999500. Expected  500 Hz audio, measured  720 Hz audio, so tuning is off by  220 Hz =  0.76 ppm
pi@WsprdaemonPi:~/soapy-test $

===============

For comparison, I plugged the same RTL-SDR+GPSDO into a second Pi with and recorded at the same set of frequencies using CubicSDR.  The audio produced by Cubic is consistently 64 or 65 Hz below the expected audio, e.g. tuned 1000 Hz below 288Mhz I measure a 935 Hz audio tone.

Expected               Measured
1500                      1540
1200                      1241
1100                      1141
1050                      1091
1020                      1061
1010                      1051
1005                      1046
1002                      1043
1000                      1041
 998                      1039
 995                      1036
 990                      1031
 980                      1021
 950                       991
 900                       942
 800                       842
 500                       542

So while Cubic is tuning 41/42 Hz low and thus generating audio which is 41 or 42 Hz too high, its error is constant and thus can be adjusted for when tuning.

  

=============

pi@WsprdaemonPi:~/soapy-test $ ./sdr-test.sh
Mon 18 Jan 2021 04:00:15 PM PST: check_tuning_accuracy() test device #1 at 287999000, expecting 1000 Hz audio at that center tuning frequency
Mon 18 Jan 2021 04:00:22 PM PST: check_tuning_accuracy() device #1 tuned to frequency 287998500. Expected 1500 Hz audio, measured 1556 Hz audio, so tuning is off by   56 Hz =  0.19 ppm
[... cut ...]
Mon 18 Jan 2021 04:01:36 PM PST: check_tuning_accuracy() device #1 tuned to frequency 287999100. Expected  900 Hz audio, measured 1145 Hz audio, so tuning is off by  245 Hz =  0.85 ppm
Mon 18 Jan 2021 04:01:43 PM PST: check_tuning_accuracy() device #1 tuned to frequency 287999200. Expected  800 Hz audio, measured  934 Hz audio, so tuning is off by  134 Hz =  0.46 ppm
Mon 18 Jan 2021 04:01:50 PM PST: check_tuning_accuracy() device #1 tuned to frequency 287999500. Expected  500 Hz audio, measured  720 Hz audio, so tuning is off by  220 Hz =  0.76 ppm
pi@WsprdaemonPi:~/soapy-test $

righthalfplane, 2021-01-19:
Hi Rob,

There is a new version of sdrTest. A few changes from SdrGlut were missing. I tested it with a RSP2 and the RspDuo and they only drifted 1 HZ while testing and were linear to the frequency shifts. The RTL-Stick drifted 16 HZ while testing - so its hard to tell how linear it is.

rrobinett, 2021-01-19:
Hi Dale,

Your changes made a big improvement in tuning accuracy.  This morning I was looking at your code and noticed that you were storing the tuning and center frequencies as floats, while the Soapy API expected doubles, so there must have been an implicit conversion from float to double which might have introduced errors.

However, in expanding my test SW I noticed that I had been changing both the tuning and center frequencies.  So I modified my code to run one test where I keep the tuning frequency constant and vary the center, and a second test where I keep the center constant and vary the tuning.  As you can see below, when I change only the center frequency there are errors up to 430 Hz.  With a constant center the errors are less than 40 Hz.

These results suggest to me that there may be a problem in the Soapy library.  

Thanks,

Rob

=========

pi@WsprdaemonPi:~/soapy-test $ ./sdr-test.sh
Tue 19 Jan 2021 12:19:44 PM PST: check_tuning_accuracy() with signal at 576000000, tune device #1 around  575999000, expecting 1000 Hz audio at the center tuning

Constant tuning which generates some big error):
Tue 19 Jan 2021 12:19:51 PM PST: check_tuning_accuracy() device #1 tuned to 575999000, center 575899500. Expected 1000 Hz audio, measured  966 Hz audio, so tuning is off by  -34 Hz = -0.05 ppm
Tue 19 Jan 2021 12:19:59 PM PST: check_tuning_accuracy() device #1 tuned to 575999000, center 575899800. Expected 1000 Hz audio, measured 1253 Hz audio, so tuning is off by  253 Hz =  0.43 ppm
Tue 19 Jan 2021 12:20:07 PM PST: check_tuning_accuracy() device #1 tuned to 575999000, center 575899900. Expected 1000 Hz audio, measured 1347 Hz audio, so tuning is off by  347 Hz =  0.60 ppm
Tue 19 Jan 2021 12:20:14 PM PST: check_tuning_accuracy() device #1 tuned to 575999000, center 575899950. Expected 1000 Hz audio, measured 1393 Hz audio, so tuning is off by  393 Hz =  0.68 ppm
Tue 19 Jan 2021 12:20:22 PM PST: check_tuning_accuracy() device #1 tuned to 575999000, center 575899980. Expected 1000 Hz audio, measured 1421 Hz audio, so tuning is off by  421 Hz =  0.73 ppm
Tue 19 Jan 2021 12:20:30 PM PST: check_tuning_accuracy() device #1 tuned to 575999000, center 575899990. Expected 1000 Hz audio, measured 1430 Hz audio, so tuning is off by  430 Hz =  0.74 ppm
Tue 19 Jan 2021 12:20:37 PM PST: check_tuning_accuracy() device #1 tuned to 575999000, center 575900000. Expected 1000 Hz audio, measured 1024 Hz audio, so tuning is off by   24 Hz =  0.04 ppm
Tue 19 Jan 2021 12:20:45 PM PST: check_tuning_accuracy() device #1 tuned to 575999000, center 575900010. Expected 1000 Hz audio, measured 1034 Hz audio, so tuning is off by   34 Hz =  0.05 ppm
Tue 19 Jan 2021 12:20:52 PM PST: check_tuning_accuracy() device #1 tuned to 575999000, center 575900020. Expected 1000 Hz audio, measured 1044 Hz audio, so tuning is off by   44 Hz =  0.07 ppm
Tue 19 Jan 2021 12:21:00 PM PST: check_tuning_accuracy() device #1 tuned to 575999000, center 575900050. Expected 
1000 Hz audio, measured 1073 Hz audio, so tuning is off by   73 Hz =  0.12 ppm
Tue 19 Jan 2021 12:21:08 PM PST: check_tuning_accuracy() device #1 tuned to 575999000, center 575900100. Expected 1000 Hz audio, measured 1120 Hz audio, so tuning is off by  120 Hz =  0.20 ppm
Tue 19 Jan 2021 12:21:15 PM PST: check_tuning_accuracy() device #1 tuned to 575999000, center 575900200. Expected 1000 Hz audio, measured 1215 Hz audio, so tuning is off by  215 Hz =  0.37 ppm
Tue 19 Jan 2021 12:21:23 PM PST: check_tuning_accuracy() device #1 tuned to 575999000, center 575900500. Expected 1000 Hz audio, measured 1083 Hz audio, so tuning is off by   83 Hz =  0.14 ppm

Constant center with changes in tuning frequency:
Tue 19 Jan 2021 12:21:30 PM PST: check_tuning_accuracy() device #1 tuned to 575998500, center 575900000. Expected 1500 Hz audio, measured 1495 Hz audio, so tuning is off by   -5 Hz =  0.00 ppm
Tue 19 Jan 2021 12:21:38 PM PST: check_tuning_accuracy() device #1 tuned to 575998800, center 575900000. Expected 1200 Hz audio, measured 1215 Hz audio, so tuning is off by   15 Hz =  0.02 ppm
Tue 19 Jan 2021 12:21:46 PM PST: check_tuning_accuracy() device #1 tuned to 575998900, center 575900000. Expected 1100 Hz audio, measured 1120 Hz audio, so tuning is off by   20 Hz =  0.03 ppm
Tue 19 Jan 2021 12:21:53 PM PST: check_tuning_accuracy() device #1 tuned to 575998950, center 575900000. Expected 1050 Hz audio, measured 1073 Hz audio, so tuning is off by   23 Hz =  0.03 ppm
Tue 19 Jan 2021 12:22:01 PM PST: check_tuning_accuracy() device #1 tuned to 575998980, center 575900000. Expected 1020 Hz audio, measured 1044 Hz audio, so tuning is off by   24 Hz =  0.04 ppm
Tue 19 Jan 2021 12:22:09 PM PST: check_tuning_accuracy() device #1 tuned to 575998990, center 575900000. Expected 1010 Hz audio, measured 1034 Hz audio, so tuning is off by   24 Hz =  0.04 ppm
Tue 19 Jan 2021 12:22:16 PM PST: check_tuning_accuracy() device #1 tuned to 575999000, center 575900000. Expected 1000 Hz audio, measured 1025 Hz audio, so tuning is off by   25 Hz =  0.04 ppm
Tue 19 Jan 2021 12:22:24 PM PST: check_tuning_accuracy() device #1 tuned to 575999010, center 575900000. Expected  990 Hz audio, measured 1015 Hz audio, so tuning is off by   25 Hz =  0.04 ppm
Tue 19 Jan 2021 12:22:31 PM PST: check_tuning_accuracy() device #1 tuned to 575999020, center 575900000. Expected  980 Hz audio, measured 1005 Hz audio, so tuning is off by   25 Hz =  0.04 ppm
[... cut ...]
Tue 19 Jan 2021 12:22:46 PM PST: check_tuning_accuracy() device #1 tuned to 575999100, center 575900000. Expected  900 Hz audio, measured  928 Hz audio, so tuning is off by   28 Hz =  0.04 ppm
Tue 19 Jan 2021 12:22:54 PM PST: check_tuning_accuracy() device #1 tuned to 575999200, center 575900000. Expected  800 Hz audio, measured  831 Hz audio, so tuning is off by   31 Hz =  0.05 ppm
Tue 19 Jan 2021 12:23:02 PM PST: check_tuning_accuracy() device #1 tuned to 575999500, center 575900000. Expected  500 Hz audio, measured  538 Hz audio, so tuning is off by   38 Hz =  0.06 ppm
pi@WsprdaemonPi:~/soapy-test $

righthalfplane, 2021-01-19:
Yes, the floats were the main problem.
 
Fc needs to be kept no further than  (+ or -) 0.5*samplerate (the response often drops at the edge so you should probably keep it down to 0.25*samplerate) from f.  Changing fc in that range should not change the output. I tried it with my RSPduo and changed the fc about (+ and -)  0.8*(0.5*samplerate) and leaving f the same - the output did not change.

rrobinett, 2021-01-20:
Limiting Fc to less than 50% would of course make sense.  However in my latest test trying 1/2/5/10/50% of sample rate (10000/20000...1000000 Hz of the default 2 Msps sample rate) as offset didn't help.  The errors bounced from -74 to +327 Hz.

I am now running a very long test where it increment the Fc 1010 Hz from 100350 to 101360 Hz.  So far the errors increment by 1 Hz.  It will be sufficient if I can find a magic Fc offset that gives me 0 or the same error at all tuning frequencies. Hope springs eternal...

Rob

=======

pi@WsprdaemonPi:~/soapy-test $ ./sdr-test.sh
Tue 19 Jan 2021 04:36:34 PM PST: check_tuning_accuracy() with signal at 576000000, tune device #1 around  575999000, expecting 1000 Hz audio at the center tuning
Tue 19 Jan 2021 04:36:42 PM PST: check_tuning_accuracy() device #1 tuned to 575999000, center 575979000. Expected 1000 Hz audio, measured  926 Hz audio, so tuning is off by  -74 Hz = -0.12 ppm
Tue 19 Jan 2021 04:36:50 PM PST: check_tuning_accuracy() device #1 tuned to 575999000, center 575959000. Expected 1000 Hz audio, measured 1133 Hz audio, so tuning is off by  133 Hz =  0.23 ppm
Tue 19 Jan 2021 04:36:57 PM PST: check_tuning_accuracy() device #1 tuned to 575999000, center 575899000. Expected 1000 Hz audio, measured 1327 Hz audio, so tuning is off by  327 Hz =  0.56 ppm
Tue 19 Jan 2021 04:37:05 PM PST: check_tuning_accuracy() device #1 tuned to 575999000, center 575799000. Expected 1000 Hz audio, measured 1096 Hz audio, so tuning is off by   96 Hz =  0.16 ppm
Tue 19 Jan 2021 04:37:13 PM PST: check_tuning_accuracy() device #1 tuned to 575999000, center 575599000. Expected 1000 Hz audio, measured 1049 Hz audio, so tuning is off by   49 Hz =  0.08 ppm
Tue 19 Jan 2021 04:37:20 PM PST: check_tuning_accuracy() device #1 tuned to 575999000, center 574999000. Expected 1000 Hz audio, measured 1327 Hz audio, so tuning is off by  327 Hz =  0.56 ppm
pi@WsprdaemonPi:~/soapy-test $

rrobinett, 2021-01-20:
Hi Dale,

I have just completed a test of listening to a 567,000,000 carrier while tuning to 575,999,000.  So I expect a 1000 Hz audio tone.
In the attached file I change only the center frequency in 1 Hz steps from (567,000,000 - 100340 )to (567,000,000 - 101350).  The resulting audio error starts goes in 1 Hz steps from:  +4 ... -152,  (then jumps +421 to)+269...+32,
(then jumps +414 to)+446...-26,  (then jumps +412 to) +390..297.

The +412/414/420 jumps suggest to me there is still some rounding or number format error lurking in the SW.  As a next step I can try loading the Soapy driver for my SDRPlay and see if it exhibits similar behavior.

Thanks for your attention,

Rob
[offset_scan.log](https://github.com/righthalfplane/SdrGlut/files/5840016/offset_scan.log)

righthalfplane, 2021-01-20:
I went from fc 602.307440e6 to fc 602.309340e6 in 100 hz steps about f 602.308440e6. looks good to me with my RSPduo

test00.txt:Rough   frequency:         1198
test01.txt:Rough   frequency:         1197
test02.txt:Rough   frequency:         1197
test03.txt:Rough   frequency:         1198
test04.txt:Rough   frequency:         1198
test05.txt:Rough   frequency:         1199
test06.txt:Rough   frequency:         1197
test07.txt:Rough   frequency:         1198
test08.txt:Rough   frequency:         1198
test09.txt:Rough   frequency:         1199
test10.txt:Rough   frequency:         1199
test11.txt:Rough   frequency:         1198
test12.txt:Rough   frequency:         1198
test13.txt:Rough   frequency:         1199
test14.txt:Rough   frequency:         1199
test15.txt:Rough   frequency:         1200
test16.txt:Rough   frequency:         1198
test17.txt:Rough   frequency:         1198
test18.txt:Rough   frequency:         1199
test19.txt:Rough   frequency:         1199

rrobinett, 2021-01-20:
Clint KA7OEI suggested that the 'center freq' offset errors are intrinsic to the chip design, but the drive can be compiled to print that error as part of tuning.  That error could then be incorporated into the 'tuning freq' given the chip.

However I can't find that flag in ~/rtl-sdr/src/tuner_r82xx.c, so I'm blocked at this step.

righthalfplane, 2021-01-21:
Have you tried RTL with a 1 PPM TCXO - they are not very expensive.

rrobinett, 2021-01-21:
I have been using a RTL-SDR V3 modified to accept an external GPSDO clock which is accurate to 10e-12.

Clint KA7OEI and I worked on this tuning error for several hours this afternoon and observed that the rtl_fm utility exhibits the similar tuning problems.
In the version of librtlsdr at https://github.com/pothosware/librtlsdr we found in "tuner_r82xx.c" the define for 
PRINT_ACTUAL_VCO_AND_ERR which when set to 1 causes it to print the tuning error associated with  the settings of the RA802 tuner
From our investigations it seems that tuner error frequency is not incorporated into the final tuning of the RTL-SDR, nor is it communicated to Soapy.
So it appears there is a tuning error in rtl-sdr which propagates up to Soapy and then to your program. 
Thus this appears to be a bug unique to the RTL-SDR driver and I am going to communicate the problem to the https://github.com/pothosware/librtlsdr author.
Thanks for your support on this, but I think the tuning problem should be corrected below your SW.

righthalfplane, 2021-01-21:
Good Luck ! Bye !

closed 2021-01-21 by righthalfplane

rrobinett, 2021-01-22:
Thanks for everything !