righthalfplane/SdrGlut #20: SdrGlut doesn't see SDRPlay

rrobinett, 2021-01-26:
Hi Dale, 

I'm sorry to bother you, but sdrglut.x crashes when I select the RTL-SDR on my Pi 4 after a clean installation of the OS and Soapy.

```
pi@wsprdaemon-pi:~/SdrGlut $ ./sdrglut.x 
..........
tuner=Rafael Micro R820T
Found Rafael Micro R820T tuner
device 0xc40da8
receive channels 1 transmit channels 0
Hardware origin: https://github.com/pothosware/SoapyRTLSDR
Hardware rtl: 0
cMalloc Out of Memory Requested (-1435816424) Total Used (1430890228) tag (8891)
cMalloc Out of Memory Requested (-1435816424) Total Used (1430890228) tag (8891)
cMalloc Out of Memory Requested (-1435816424) Total Used (1430890228) tag (8892)
cMalloc Out of Memory Requested (-1435816424) Total Used (1430890228) tag (8892)
Segmentation fault
pi@wsprdaemon-pi:~/SdrGlut $ 
```
I am also unable to get SdrGlut to recognize the SDRPlay attached to my Pi 4.
lsusb lists the RP2, although I don't understand why it doesn't print out it name since the device ID 1df7:3010 has been entered into /var/lib/usbutils/usb.ids:

```
pi@wsprdaemon-pi:~/SdrGlut/sdrTest $ SoapySDRUtil --find
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Found Rafael Micro R820T tuner
Found device 0
  driver = rtlsdr
  label = Generic RTL2832U OEM :: 00000001
  manufacturer = Realtek
  product = RTL2838UHIDIR
  serial = 00000001
  tuner = Rafael Micro R820T

Found device 1
  driver = sdrplay
  label = SDRplay Dev0 RSP2 1710047820
pi@wsprdaemon-pi:~/sdrplay $ grep "1df7\|3010.*RSP" /var/lib/usbutils/usb.ids
1df7  SDRplay
	3010  RSP2/RSP2pro
pi@wsprdaemon-pi:~/sdrplay $ lsusb
Bus 002 Device 002: ID 05e3:0749 Genesys Logic, Inc.
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 010: ID 1df7:3010
Bus 001 Device 007: ID 1c4f:0048 SiGma Micro
Bus 001 Device 006: ID 046e:6000 Behavior Tech. Computer Corp.
Bus 001 Device 004: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 001 Device 003: ID 0bda:2838 Realtek Semiconductor Corp. RTL2838 DVB-T
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
pi@wsprdaemon-pi:~/sdrplay $
```

Finally, I am not able to compile sdrTest with or without #define LIQUID_VERSION_4 1:

```
[pi@wsprdaemon-pi:~/SdrGlut/sdrTest $ g++ -O2 -std=c++11 -Wno-deprecated -o sdrTest sdrTest.cpp mThread.cpp cMalloc.c -lrtaudio -lSoapySDR -lliquid -lpthread -lopenal
sdrTest.cpp: In function ‘int setFilters(playData*, Filters*)’:
sdrTest.cpp:1378:45: error: invalid conversion from ‘int’ to ‘liquid_ampmodem_type’ [-fpermissive]
     f->demodAM = ampmodem_create(0.5, mode, iflag);
                                             ^~~~~
sdrTest.cpp:1378:50: error: too few arguments to function ‘ampmodem_s* ampmodem_create(float, float, liquid_ampmodem_type, int)’
     f->demodAM = ampmodem_create(0.5, mode, iflag);
                                                  ^
In file included from sdrTest.cpp:9:
/usr/include/liquid/liquid.h:5906:10: note: declared here
 ampmodem ampmodem_create(float _m,
          ^~~~~~~~~~~~~~~
pi@wsprdaemon-pi:~/SdrGlut/sdrTest $
```

I have made every effort to start with a clean OS and add only the Soapy programs and associated drivers, and SoapyUtil will list both the RTL-SDR and the RP2 and give me accurate detailed descriptions of their capabilities:

```
pi@wsprdaemon-pi:~/SdrGlut/sdrTest $ SoapySDRUtil --find
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Found Rafael Micro R820T tuner
Found device 0
  driver = rtlsdr
  label = Generic RTL2832U OEM :: 00000001
  manufacturer = Realtek
  product = RTL2838UHIDIR
  serial = 00000001
  tuner = Rafael Micro R820T

Found device 1
  driver = sdrplay
  label = SDRplay Dev0 RSP2 1710047820

pi@wsprdaemon-pi:~/SdrGlut/sdrTest $
```

Since Soapyutil runs, I think my problems are with my SdrGlut build.  It built and ran on another Pi4, but it had many other programs installed like CubicSDR. This Pi 4 is as clean as I can make it, but I seem to have missed some steps needed in its setup.

thanks

righthalfplane, 2021-01-26:
*****
Finally, I am not able to compile sdrTest with or without #define LIQUID_VERSION_4 1:
It shoud be - 
******
#define LIQUID_VERSION_4 1

not 

#define LIQUID_VERSION_4 1:

Try  that first.

rrobinett, 2021-01-26:
I can compile sdrTest, but it segfaults when I run it, just like SdrGlut
does:

pi@wsprdaemon-pi:~/SdrGlut/sdrTest $ g++ -O2 -std=c++11 -Wno-deprecated -o
sdrTest sdrTest.cpp mThread.cpp cMalloc.c -lrtaudio -lSoapySDR -lliquid
-lpthread -lopenal
pi@wsprdaemon-pi:~/SdrGlut/sdrTest $ SoapySDRUtil --find
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Found Rafael Micro R820T tuner
Found device 0
  driver = rtlsdr
  label = Generic RTL2832U OEM :: 00000001
  manufacturer = Realtek
  product = RTL2838UHIDIR
  serial = 00000001
  tuner = Rafael Micro R820T

Found device 1
  driver = sdrplay
  label = SDRplay Dev0 RSP2 1710047820

pi@wsprdaemon-pi:~/SdrGlut/sdrTest $ ./sdrTest -f 575999000 -fc 575000000
-usb -device 0 -faudio 120000 -timeout 3 -file sdraudio.raw
fc 575000000.000000 f 575999000.000000 dt 5e-07 samplerate 2000000
[ERROR] SoapySSDPEndpoint::sendTo(udp://[ff02::c]:1900) = -1
  sendto(udp://[ff02::c]:1900) [99: Cannot assign requested address]
Found Rafael Micro R820T tuner
[INFO] [UHD] linux; GNU C++ version 8.2.0; Boost_106700; UHD_3.13.1.0-3
Number of Devices Found: 2
SDR device = 0    label = default
SDR device = 1    label = Generic RTL2832U OEM :: 00000001

device = 0 selected
default_input = False
default_output = False
device_id = 2
driver = audio
label = default

Rx antennas:
RX

Setting Info:

Segmentation fault
pi@wsprdaemon-pi:~/SdrGlut/sdrTest $

righthalfplane, 2021-01-27:
You need to use  -device 1 - that is the RTL2832U.

SDR device = 0    label = default
SDR device = 1    label = Generic RTL2832U OEM :: 00000001

rrobinett, 2021-01-27:
I tried device 0 and 1 and got a seg fault on both.  device =2 gave me the
expected 'no such device'

righthalfplane, 2021-01-27:
Show me the full output of

SoapySDRUtil --info

and the full output of   -

./sdrTest -f 575999000 -fc 575000000 -usb -device 1 -faudio 120000 -timeout 3 -file sdraudio.raw

and the full output of SdrGlut when you start it up and select Device "Generic RTL2832..."

rrobinett, 2021-01-27:
Thanks for your attention.

***************************************************************************************************
pi@wsprdaemon-pi:~/SdrGlut/sdrTest $ SoapySDRUtil --info
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Lib Version: v0.8.0-g351896d3
API Version: v0.8.0
ABI Version: v0.8
Install root: /usr/local
Search path:  /usr/local/lib/SoapySDR/modules0.8
Module found: /usr/local/lib/SoapySDR/modules0.8/librtlsdrSupport.so
 (0.3.2-b5489df)
Module found: /usr/local/lib/SoapySDR/modules0.8/libsdrPlaySupport.so
(0.3.0-208f95d)
Available factories... rtlsdr, sdrplay
Available converters...
 -  CF32 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS16 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS32 -> [CS32]
 -   CS8 -> [CF32, CS16, CS8, CU16, CU8]
 -  CU16 -> [CF32, CS16, CS8]
 -   CU8 -> [CF32, CS16, CS8]
 -   F32 -> [F32, S16, S8, U16, U8]
 -   S16 -> [F32, S16, S8, U16, U8]
 -   S32 -> [S32]
 -    S8 -> [F32, S16, S8, U16, U8]
 -   U16 -> [F32, S16, S8]
 -    U8 -> [F32, S16, S8]

***************************************************************************************************
pi@wsprdaemon-pi:~/SdrGlut/sdrTest $ ./sdrTest -f 575999000 -fc 575000000
-usb -device 1 -faudio 120000 -timeout 3 -file sdraudio.raw
fc 575000000.000000 f 575999000.000000 dt 5e-07 samplerate 2000000
[ERROR] SoapySSDPEndpoint::sendTo(udp://[ff02::c]:1900) = -1
  sendto(udp://[ff02::c]:1900) [99: Cannot assign requested address]
Found Rafael Micro R820T tuner
[INFO] [UHD] linux; GNU C++ version 8.2.0; Boost_106700; UHD_3.13.1.0-3
Number of Devices Found: 2
SDR device = 0    label = default
SDR device = 1    label = Generic RTL2832U OEM :: 00000001

device = 1 selected
available = Yes
driver = rtlsdr
label = Generic RTL2832U OEM :: 00000001
manufacturer = Realtek
product = RTL2838UHIDIR
rtl = 0
serial = 00000001
tuner = Rafael Micro R820T

Found Rafael Micro R820T tuner

Rx antennas:
RX

Setting Info:

Segmentation fault

***************************************************************************************************
pi@wsprdaemon-pi:~/SdrGlut $ ./sdrglut.x
LIQUID_VERSION_NUMBER 1003000
Lib Version: v0.6.1-4+b4
API Version: v0.6.0
ABI Version: v0.6
Install root: /usr
Search path: /usr/lib/arm-linux-gnueabihf/SoapySDR/modules0.6
Search path: /usr/local/lib/arm-linux-gnueabihf/SoapySDR/modules0.6
Search path: /usr/local/lib/SoapySDR/modules0.6
Module found:
/usr/lib/arm-linux-gnueabihf/SoapySDR/modules0.6/libHackRFSupport.so
Module found:
/usr/lib/arm-linux-gnueabihf/SoapySDR/modules0.6/libLMS7Support.so
Module found:
/usr/lib/arm-linux-gnueabihf/SoapySDR/modules0.6/libRedPitaya.so
Module found:
/usr/lib/arm-linux-gnueabihf/SoapySDR/modules0.6/libairspySupport.so
Module found:
/usr/lib/arm-linux-gnueabihf/SoapySDR/modules0.6/libaudioSupport.so
Module found:
/usr/lib/arm-linux-gnueabihf/SoapySDR/modules0.6/libbladeRFSupport.so
Module found:
/usr/lib/arm-linux-gnueabihf/SoapySDR/modules0.6/libosmosdrSupport.so
Module found:
/usr/lib/arm-linux-gnueabihf/SoapySDR/modules0.6/libremoteSupport.so
Module found:
/usr/lib/arm-linux-gnueabihf/SoapySDR/modules0.6/librtlsdrSupport.so
Module found:
/usr/lib/arm-linux-gnueabihf/SoapySDR/modules0.6/libuhdSupport.so
[ERROR] SoapySSDPEndpoint::sendTo(udp://[ff02::c]:1900) = -1
  sendto(udp://[ff02::c]:1900) [99: Cannot assign requested address]
Found Rafael Micro R820T tuner
[INFO] [UHD] linux; GNU C++ version 8.2.0; Boost_106700; UHD_3.13.1.0-3
default_input=False
default_output=False
device_id=2
driver=audio
label=default
available=Yes
driver=rtlsdr
label=Generic RTL2832U OEM :: 00000001
manufacturer=Realtek
product=RTL2838UHIDIR
rtl=0
serial=00000001
tuner=Rafael Micro R820T
Found Rafael Micro R820T tuner
receive channels 1 transmit channels 0
available=Yes
driver=rtlsdr
label=Generic RTL2832U OEM :: 00000001
manufacturer=Realtek
product=RTL2838UHIDIR
rtl=0
serial=00000001
tuner=Rafael Micro R820T
Found Rafael Micro R820T tuner
device 0x1b353e0
receive channels 1 transmit channels 0
Hardware origin: https://github.com/pothosware/SoapyRTLSDR
Hardware rtl: 0
cMalloc Out of Memory Requested (-1441088048) Total Used (1428254416) tag
(8891)
cMalloc Out of Memory Requested (-1441088048) Total Used (1428254416) tag
(8891)
cMalloc Out of Memory Requested (-1441088048) Total Used (1428254416) tag
(8892)
cMalloc Out of Memory Requested (-1441088048) Total Used (1428254416) tag
(8892)
Segmentation fault
pi@wsprdaemon-pi:~/SdrGlut $

righthalfplane, 2021-01-27:
As I thought, you have two version of soapy installed. Uninstall one of them -

sudo apt-get remove libsoapysdr0.6
sudo apt-get remove libsoapysdr-dev

and rebuild sdrglut.

rrobinett, 2021-01-28:
Thanks, that fixed it.
I don't understand how that wrong version was loaded.
It seems I have all I need to finish.

righthalfplane, 2021-01-28:
Let me know when you get it finished.

closed 2021-01-28 by righthalfplane