righthalfplane/SdrGlut #23: can't run on ubuntu 19.04

luoGanda, 2021-05-22:
I put program in /opt/SdrGlut.

1) clicking runSdrGlut.sh: nothing happens

2) running runSdrGlut.sh in terminal:
  *** stack smashing detected ***: <unknown> terminated
  ./runSdrGlut.sh: line 4:  2503 Aborted                 (core dumped) ./sdrglut.x

3) running ./sdrglut.x:
LIQUID_VERSION_NUMBER 1003000
Lib Version: v0.6.1-4
API Version: v0.6.0
ABI Version: v0.6
Install root: /usr
Search path: /usr/lib/x86_64-linux-gnu/SoapySDR/modules0.6
Search path: /usr/local/lib/x86_64-linux-gnu/SoapySDR/modules0.6
Search path: /usr/local/lib/SoapySDR/modules0.6
Module found: /usr/lib/x86_64-linux-gnu/SoapySDR/modules0.6/libaudioSupport.so
Module found: /usr/lib/x86_64-linux-gnu/SoapySDR/modules0.6/libmiriSupport.so
Module found: /usr/lib/x86_64-linux-gnu/SoapySDR/modules0.6/libosmosdrSupport.so
Module found: /usr/lib/x86_64-linux-gnu/SoapySDR/modules0.6/librtlsdrSupport.so
**If i use here rtlsdr it works ok**, but with sdrplay rsp1 it hangs when clicking on device.

4) setting export LD_LIBRARY_PATH=. in terminal and running ./runSdrGlut.sh or ./sdrglut.x gives same result as above in 2)

5) unsetting LD_LIBRARY_PATH and setting export SOAPY_SDR_ROOT=.
  * running ./runSdrGlut.sh gives same stack result as above in 2)
  * running ./sdrglut.x gives:
LIQUID_VERSION_NUMBER 1003000
Lib Version: v0.6.1-4
API Version: v0.6.0
ABI Version: v0.6
Install root: .
Search path: ./lib/x86_64-linux-gnu/SoapySDR/modules0.6
No modules found!

~~~~~~~~~~~~~~
By the way - i also tried to compile it - but i get this error:
/usr/bin/ld: ./Radio/Radio.o: in function `doWindow(double*, double*, long, int) [clone .part.22]':
Radio.cpp:(.text+0x1277): undefined reference to `liquid_blackmanharris'
/usr/bin/ld: Radio.cpp:(.text+0x1377): undefined reference to `liquid_blackmanharris7'
/usr/bin/ld: Radio.cpp:(.text+0x1417): undefined reference to `liquid_hann'
/usr/bin/ld: Radio.cpp:(.text+0x1457): undefined reference to `liquid_hamming'
/usr/bin/ld: Radio.cpp:(.text+0x1497): undefined reference to `liquid_flattop'
collect2: error: ld returned 1 exit status
make: *** [makefileUbunta:107: fireball] Error 1

righthalfplane, 2021-05-23:
The built version only runs on uBuntu 18.04 as indicated. It appears that uBuntu 19.04 is different from uBuntu 18.04 and uBuntu 20.04.  Some how you have loaded the wrong version version of libliquid - you must have two versions installed.

You can fix the problem in one of two ways -

1). Force load the version of libliquid that is in /usr/local/lib.

 2). Uninstall the extra version of libliquid (the one in /usr/local/lib including the include file). You will then get a compilation error that can be corrected by a change in the file MainSource/firstFile.h comment out the line -

//#define WINDOWS_LONG_NAMES 1

then it should then build.

luoGanda, 2022-04-08:
Yes, i had two versions installed - i've got rid of newer one(1.3.2, i think it was libliquid2d),
so now i have only one - libliquid1d(v1.3.1). There were also two drivers version of sdrplay, i've got rid of 3.xx and only 2.xx is there.

SdrGlut compiled, but now it's seg fault:

anon@anon:~/packs/SdrGlut$ ./sdrglut.x 
LIQUID_VERSION_NUMBER 1003000
Lib Version: v0.6.1-4
API Version: v0.6.0
ABI Version: v0.6
Install root: /usr
Search path: /usr/lib/x86_64-linux-gnu/SoapySDR/modules0.6
Search path: /usr/local/lib/x86_64-linux-gnu/SoapySDR/modules0.6
Search path: /usr/local/lib/SoapySDR/modules0.6
Module found: /usr/lib/x86_64-linux-gnu/SoapySDR/modules0.6/libHackRFSupport.so
Module found: /usr/lib/x86_64-linux-gnu/SoapySDR/modules0.6/libLMS7Support.so
Module found: /usr/lib/x86_64-linux-gnu/SoapySDR/modules0.6/libRedPitaya.so
Module found: /usr/lib/x86_64-linux-gnu/SoapySDR/modules0.6/libairspySupport.so
Module found: /usr/lib/x86_64-linux-gnu/SoapySDR/modules0.6/libaudioSupport.so
Module found: /usr/lib/x86_64-linux-gnu/SoapySDR/modules0.6/libbladeRFSupport.so
Module found: /usr/lib/x86_64-linux-gnu/SoapySDR/modules0.6/libmiriSupport.so
Module found: /usr/lib/x86_64-linux-gnu/SoapySDR/modules0.6/libosmosdrSupport.so
Module found: /usr/lib/x86_64-linux-gnu/SoapySDR/modules0.6/libremoteSupport.so
Module found: /usr/lib/x86_64-linux-gnu/SoapySDR/modules0.6/librtlsdrSupport.so
Module found: /usr/lib/x86_64-linux-gnu/SoapySDR/modules0.6/libuhdSupport.so

RtApiAlsa::getDeviceInfo: snd_pcm_open error for device (hw:0,0), Device or resource busy.

[INFO] [UHD] linux; GNU C++ version 8.2.0; Boost_106700; UHD_3.13.1.0-3
default_input=True
default_output=False
device_id=0
driver=audio
default_input=False
default_output=False
device_id=1
driver=audio
default_input=False
default_output=False
device_id=2
driver=audio
driver=miri
label=Mirics MSi2500 default (e.g. VTX3D card)
miri=0
Using device #0: Mirics MSi2500 default (e.g. VTX3D card)
_mirisdr_alloc_async_buffers
Lost samples!
b6 de 0e 00 28 30 00 00 7f 80 31 9b df f8 f7 4c 
Lost samples!
OOOOOOOOOOOOOOOOOOOOSegmentation fault (core dumped)
anon@anon:~/packs/SdrGlut$ SoapySDRUtil --find
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

libusb: error [udev_hotplug_event] ignoring udev action bind
Found device 0
  default_input = True
  default_output = True
  device_id = 0
  driver = audio
  label = hw:HDA Intel,0

Found device 1
  default_input = False
  default_output = False
  device_id = 1
  driver = audio
  label = hw:HDA Intel,2

Found device 2
  default_input = False
  default_output = False
  device_id = 2
  driver = audio
  label = default

Found device 3
  driver = sdrplay
  label = SDRplay Dev0 RSP1 ...
~~~~~

In another linux version(i think debian 10/buster) app starts,
but when i press button after Radio and then sdrplay - only ooooooooooo are in command line and nothing happens(spectrum window is not opened).

Both was git clone'd from afresh.