righthalfplane/SdrGlut #26: RedPitaya Support

effeerre75, 2021-08-25:
Hi, using it on Mac Catalina. Works fine with RTLSDR. 
Just tried to add support for my RedPitaya adding the module SopayRedPitaya. 
I compiled it with macports and added to the folder "modules0.8" inside SDRglut package.
At the start it find the module but not the board.
I think I'm missing something! 😃
Thanks and have a good day!

[image]

righthalfplane, 2021-08-30:
@effeerre75 - I added the device string to SdrGlut. Set it to -

driver=redpitaya, addr=192.168.1.100

and it should find the redpitaya. It will be in the next version of SdrGlut or you can get it now by building SdrGlut from the source at -

https://github.com/righthalfplane/SdrGlut

effeerre75, 2021-09-05:
Had time to make a quick check! Worked fine! Thanks!

[image]
[image]