righthalfplane/SdrGlut #28: sdrplay not found - windows7

luoGanda, 2022-04-06:
On the pic there are two version of SoapySdrUtil,
left is from SdrGlut-823-Windows.zip and right is from SdrGlut-960-Windows.zip.
All version after 823 don't find sdrplay rsp1.

[image]

righthalfplane, 2022-04-07:
If you are Using a SDRPlay device with SdrGlut-960, SDRplay_RSP_API-Windows-3.07.2.exe needs to be installed. You can get it from -
https://www.sdrplay.com/downloads/

luoGanda, 2022-04-07:
I installed sdrplay 3.07.2 driver/api, but now - down picture happens.

[image]

righthalfplane, 2022-04-08:
Everything works fine with my RSP1 on my  Windows 7 laptop. Did you do the install of the OpenAL stuff ?

Some people may need to install OpenAL on Windows. Get the two packages -
OpenAL1.1.CoreSDK
OpenAL 1.1 windows install
from "https://www.openal.org/downloads"
and install them.