righthalfplane/SdrGlut #29: SdrGlut V1.29 Can't start on MacOS Monterey - failed to open display

atlex, 2022-04-27:
```
/Applications/SdrGlut.app/Contents/MacOS/SdrGlut3.sh ; exit;

The default interactive shell is now zsh.
To update your account to use zsh, please run `chsh -s /bin/zsh`.
For more details, please visit https://support.apple.com/kb/HT208050.
(base) Alexanders-MBP:~ alex$ /Applications/SdrGlut.app/Contents/MacOS/SdrGlut3.sh ; exit;
LIQUID_VERSION_NUMBER 1003002
Lib Version: v0.8.0-g926c86d9
API Version: v0.8.0
ABI Version: v0.8
Install root: .
Search path: ./lib/SoapySDR/modules0.8
Module found: ./lib/SoapySDR/modules0.8/libHackRFSupport.so
Module found: ./lib/SoapySDR/modules0.8/libLMS7Support.so
Module found: ./lib/SoapySDR/modules0.8/libairspyhfSupport.so
Module found: ./lib/SoapySDR/modules0.8/libbladeRFSupport.so
Module found: ./lib/SoapySDR/modules0.8/libnetSDRSupport.so
Module found: ./lib/SoapySDR/modules0.8/libremoteSupport.so
Module found: ./lib/SoapySDR/modules0.8/librtlsdrSupport.so
Module found: ./lib/SoapySDR/modules0.8/libsdrPlaySupport.so
freeglut (./SdrGlut2): failed to open display ''
logout
```

$ brew install --cask xquartz
```
Warning: Cask 'xquartz' is already installed.
```

righthalfplane, 2022-04-28:
Have you tried restarting your computer ?

if you do a "printenv" you should see something like -

DISPLAY=/private/tmp/com.apple.launchd.jHfWMRdtMj/org.xquartz:0

in the list.

If you are still having problems , try uninstalling the brew version of xqaurtz and get the latest version from the website -  the brew version may be out of date.

atlex, 2022-04-28:
Hi @righthalfplane,

I have restarted my macbook and checked display.

$ printenv

```
...
DISPLAY=/private/tmp/com.apple.launchd.Ue4dW8vEWg/org.xquartz:0
...
```

and SdrGlut V1.29 has been started.

Thank you!