righthalfplane/SdrGlut #30: Install Issues for MacOS Monterey 12.4

AgileEngineeringLLC, 2022-06-24:
I get to the second step of the install, and get the following error. Any suggestions? Looks to be an X11 issue, and maybe Monterey is using XQuartz instead of X!! maybe?

(base) Dons-MacBook-Pro:~ donkelly$ brew tap dholm/homebrew-sdr
==> Tapping dholm/sdr
Cloning into '/usr/local/Homebrew/Library/Taps/dholm/homebrew-sdr'...
remote: Enumerating objects: 441, done.
remote: Total 441 (delta 0), reused 0 (delta 0), pack-reused 441
Receiving objects: 100% (441/441), 120.77 KiB | 1.49 MiB/s, done.
Resolving deltas: 100% (269/269), done.
Error: Invalid formula: /usr/local/Homebrew/Library/Taps/dholm/homebrew-sdr/multimon-ng.rb
multimon-ng: Unsupported special dependency :x11
Error: Cannot tap dholm/sdr: invalid syntax in tap!
(base) Dons-MacBook-Pro:~ donkelly$

righthalfplane, 2022-06-25:
If you want to build from scratch, clone from -

https://github.com/righthalfplane/SdrGlut

and use makefileMACINTOSH or 

https://github.com/righthalfplane/SdrGlut/tree/master/SdrGlut.xcodeproj

If you just want to install -

https://github.com/righthalfplane/SdrGlut/releases/tag/v1.30

It has the prebuilt exe for MacOS Sierra, MacOS High Sierra, MacOS Mojave, MacOS Catalina and MacOS Big Sur and a native mode version for Apple-M1 Machines. Download SdrGlut-986-Install.dmg or SdrGlut-986-M1-m-Install.dmg and install SdrGlut by dragging it to the Applications folder. On MacOS, xquartz needs to be installed (https://www.xquartz.org/releases/). If you are Using a SDRPlay device, SDRplay_RSP_API-MacOSX-3.07.3 needs to be installed. You can get it from -

https://www.sdrplay.com/downloads/

righthalfplane, 2022-07-07:
I have fixed the brew install instructions - The apple M1 machines need a few changes and a new Makefile.

righthalfplane, 2022-07-08:
Yes, Sorry about that - I put it out there a few minutes ago. I aways seem to forget something.