righthalfplane/SdrGlut #31: sdrReceive M1 mac compile errors

pmosx21, 2022-07-12:
hello, 

I am trying to build sdrReceive from source but I am getting this error: 

```
make -f makefileMACINTOSH 
c++ -std=c++11  -O2 -I/usr/include  -I/opt/X11/include -Wall -DGL_SILENCE_DEPRECATION -Wno-return-type-c-linkage -Wno-deprecated-declarations   -c -o src/cReceive.o src/cReceive.cpp
src/cReceive.cpp:22:10: fatal error: 'df.h' file not found
#include <df.h>
         ^~~~~~
1 error generated.
make: *** [src/cReceive.o] Error 1
```

I tried to compile coreutils from src but it didn't clear my error. 

Any ideas? the M1 mac is always a bummer - thanks!

righthalfplane, 2022-07-12:
The M1 Mac needs some addition stuff - I just it put out on SdrGlut -

Build and install the HDF libraries-

cd hdflib
install.txt
cd ..

Then build sdrReceive -

cd  sdrReceive
make -f makefileM1Brew -j 8

pmosx21, 2022-07-12:
the install script fails with this: 

`checking shell variables initial values... done
checking build system type... Invalid configuration `arm64-apple-darwin21.5.0': machine `arm64-apple' not recognized
configure: error: /bin/sh bin/config.sub arm64-apple-darwin21.5.0 failed
make: *** [_config] Error 1`

righthalfplane, 2022-07-14:
What version of MacOS are you running ?

I don't understand - the "install.txt" script disables fortran -

configure --prefix=/usr/local --disable-fortran --with-jpeg=/usr/local --with-zlib=/usr/local

so it should not be trying to use it.