righthalfplane/SdrGlut #32: sdrReceive.x

gaur-yogesh, 2022-09-27:
Hello, how to install sdrReceive.x in Ubuntu 18?

Thanks.

righthalfplane, 2022-09-27:
First, You install SdrGlut -

How to Install SdrGlut on Ubuntu 18.04 (a software defined radio)
https://youtu.be/LVpo78ubRiI

and then build sdrReceive.x -

cd sdrReceive
make -f makefileUbuntu -j 8
 ./sdrReceive.x -h
./sdrReceive.x -f 101.5 -fm

With uBuntu 22.04 the install of rtaudio has simplified to

sudo apt-get install librtaudio-dev

perhaps Ubuntu 18 has been updated now - give it a try.

gaur-yogesh, 2022-09-28:
on running command "make -f makefileUbuntu -j 8". Following error is encountered :

src/cReceive.cpp:22:10: fatal error: df.h: No such file or directory
   22 | #include <df.h>
      |          ^~~~~~
compilation terminated.
make: *** [<builtin>: src/cReceive.o] Error 1