righthalfplane/SdrGlut #37: Problem compiling in Debian bookworm

srs4511351, 2023-01-15:
I have a Raspberry Pi 4 running DietPi bookworm.

I installed liquid-dsp 1.5.0 from source.

freeglut3 is missing, not available in bookworm, but it does have freeglut3-dev
freeglut3-dev also installs libglut3.12 and libglut-dev
CubicSDR, which also uses freeglut works.

makefileRaspbian and makefileUbunta fail with this error (at the end of this block):
g++ -O3 -DNO_FORTRAN -Wno-deprecated-declarations -pthread -I/usr/local/hdf-64/include -I/usr/X11R6/include  -I./DrawFindPointHistory -I./MainSource -I./KdTree-routines -I./glui-routines/include -I./CExpress -I./FMRadio -I./Radio -o sdrglut.x ./MainSource/Getpalette.o ./MainSource/mThread.o ./MainSource/Scene.o  ./MainSource/Utilities.o ./DrawFindPointHistory/GridPlot.o ./CExpress/FFT2.o ./CExpress/Utilities2.o ./MainSource/BackgroundQ.o ./MainSource/cMalloc.o ./MainSource/SceneList.o ./MainSource/File8.o ./MainSource/sceneDisplay.o  ./glui-routines/algebra3.o ./glui-routines/arcball.o ./glui-routines/glui.o ./glui-routines/glui_add_controls.o ./glui-routines/glui_bitmap_img_data.o ./glui-routines/glui_bitmaps.o ./glui-routines/glui_button.o ./glui-routines/glui_checkbox.o ./glui-routines/glui_column.o ./glui-routines/glui_commandline.o ./glui-routines/glui_control.o ./glui-routines/glui_edittext.o ./glui-routines/glui_filebrowser.o ./glui-routines/glui_list.o ./glui-routines/glui_listbox.o ./glui-routines/glui_mouse_iaction.o ./glui-routines/glui_node.o ./glui-routines/glui_panel.o ./glui-routines/glui_radio.o ./glui-routines/glui_rollout.o ./glui-routines/glui_rotation.o ./glui-routines/glui_scrollbar.o ./glui-routines/glui_separator.o ./glui-routines/glui_spinner.o ./glui-routines/glui_statictext.o ./glui-routines/glui_string.o ./glui-routines/glui_textbox.o ./glui-routines/glui_translation.o	./glui-routines/glui_tree.o ./glui-routines/glui_treepanel.o	./glui-routines/glui_window.o ./glui-routines/quaternion.o	./glui-routines/tools/ppm.o ./MainSource/QuitDialog.o ./MainSource/main.o  ./MainSource/audiolib.o ./Classes/CWindow.o ./Classes/CLines.o ./Classes/CLInesPlotLabels.o ./Classes/ClinesAttributes.o ./Classes/ClinesPlotAttributes.o ./Classes/CLinesPlotBox.o ./Radio/Radio.o ./Radio/DialogRadio.o ./Radio/RaceFastsdr.o ./MainSource/DialogFileOpen.o ./SdrFile/SdrFile.o ./MainSource/DialogSave.o ./SdrFile/DialogSdrFile.o ./MainSource/DialogFolder.o ./MainSource/PlayIQ.o ./MainSource/RecordIQ.o ./MainSource/printInfo.o ./Classes/Transmit.o ./MainSource/WriteToGLUIWindow.o ./Digital/Poly.o ./Digital/BatchFile.o ./MainSource/eMemory.o  ./Radio/send.o ./Radio/WriteToWindow.o ./MainSource/dialogTime.o ./SdrFile/AudioFile.o ./SdrFile/DialogAudioFile.o -lm -L/usr/local -L/usr/X11/lib -lX11 -lpthread -lglut -lGL -lGLU -lopenal -lrtaudio -lliquid -lSoapySDR -lsndfile
/usr/bin/ld: ./MainSource/main.o: in function `control_cb(int)':
main.cpp:(.text+0x278): undefined reference to `dialogTest()'
collect2: error: ld returned 1 exit status
make: *** [makefileUbunta:108: fireball] Error 1

I don't know what is causing this.

righthalfplane, 2023-01-16:
Sorry about that - the call got left behind during testing - you can just comment out the call to dialogTest or update to the latest version of SdrGlut from GitHub where it has been removed.

srs4511351, 2023-01-16:
Thanks! That did it. It works fine now.
I thought it was the bookworm pre-release operating system.