righthalfplane/SdrGlut #40: Makefile fails on MacOS Sonoma

ifuchs, 2024-01-26:
Does anyone have this working on Sonoma? If so, can you share the makefile you used?

righthalfplane, 2024-01-26:
If you are using Brew, SdrGlut->makefileM1Brew
If you are using MacPorts, SdrGlut->makefileMACINTOSH

The release page for SdrGlut is -
https://github.com/righthalfplane/SdrGlut/releases/tag/v1.32

You will need to install quartz-
https://www.xquartz.org/releases/

After that the SdrGlut in
 
SdrGlut-1164-Install.dmg

should work.

ifuchs, 2024-01-27:
Got SDRglut running on the M2 Macbook. Tuned to a local FM broadcast station and got this:

https://github.com/righthalfplane/SdrGlut/assets/5650376/7c5bd9a5-95ff-487b-9465-1195d4f4946e

righthalfplane, 2024-01-27:
The RTL-Sticks sometimes have problems - Right Click on the window and try the various sample rates - one of the speeds usually works Ok. I will try it on my M1 and see what I get.

righthalfplane, 2024-01-27:
It works for me on the M1 for FM at all but the two slowest sample rate.