righthalfplane/SdrGlut #41: Sound errors and dropouts

srs4511351, 2024-03-09:
I am having problems with occasional dropouts in the sound accompanied by errors displayed in the terminal:
```
SDRplay Dev0 RSP1A 19110C0797 (0) count 2114 sound restarted
SDRplay Dev0 RSP1A 19110C0797 (0) count 2443 sound restarted
...
```
It didn't do this on older OS versions. I tried an old SdrGlut version and it also had the problem.
Maybe it is something to do with newer OSes or packages?

After I saw the commit for [Fixed Errors in Sound](https://github.com/righthalfplane/SdrGlut/commit/e170a78d0967a8c912d1b1112ea1670f239d9044), I hoped it had been fixed.
This time it took several minutes before the problem started. Then there was several minutes with an error every few seconds, then it was better again.
Most of the  time the CPU meter stayed at a level of about 25%, even when there were errors.

Do you know of anything I can try?

The new iqSDR does not have this problem, but the CPU load is much higher than SdrGlut.

righthalfplane, 2024-03-09:
Yes, you are right in both cases. The dropout problem only occurs on some systems with some devices. What is happening is the audio processing gets out of sync with the I/Q processing - it is a bug in the OpenAL routines - the playing of data and the freeing of buffers does not happen fast enough. I am going to try a fix in the next week or two.  Normally, iqSDR has about 20-30 percent higher CPU load - I found part of the problem, but I am still looking for the remainder.

srs4511351, 2024-03-09:
Thank you for making me feel like I'm not alone.
I look forward to a fix, but don't break your back on my account.

I have many SDR applications installed on my Raspberry Pi 4 and it seems to be typical for those who use GNU Radio to have a higher CPU load.

GdrGlut is the only one that I can use 10 MHz sample rate with minimal dropouts.

srs4511351, 2024-03-14:
I tried your latest version, 1216.
There were 2 sound restarted dropouts right after I started GdrGlut, then no more for a long time, when Firefox used a lot of CPU.
It seems to be a lot better. The real improvement is at 10 MHz sample rate. It is far better there.

righthalfplane, 2024-03-15:
With some SDR devices, you can increase the sample rate above 10 MSPS by opening the "SDR Dialog" selecting "Audio Threads" 1 or 2 or 3 and doing an "Apply" - the HackRF one will usually go up to 15 MSPS or 20 MSPS.

srs4511351, 2025-01-07:
With SdrGlut 1236 - fixed audio buffer, I get occasional errors like this:
SDRplay Dev0 RSP1A 19110C0797 (0) count 14271 sound restarted error 0

They aren't too obvious audibly.

righthalfplane, 2025-01-07:
I found and fixed a couple of errors, but I am still trying to track down some of the strange things that I am seeing.
My RSP2 is off by about 12 seconds every 20 minutes - I do not know how the RSP1A compares to that, but it may do a restart if the drift is enough. Gqrx and CubicSDR will get errors and do a restart - with out telling you and you do not notice it until the error rate gets really high.

srs4511351, 2025-01-07:
I don't understand time drift on an SDR. Can you explain?

As far as frequency accuracy, my RSP1A is typically accurate within 10 or 20 Hz.

iqSDR does not show any such errors.

righthalfplane, 2025-01-07:
When I test the SDR's, I will do about 1/2 hour recording of I/Q data, while I drive around. My last test transmission is usually at about 20 minutes. I am also timing it with a stop watch. I find that with my Sdrplay SDR's (RSP1, RSP2, RspDuo) the time is off by about 12 seconds at the 20 minute mark -  where as with my other SDR's - HackRF One, BladeRF, AntSDR e200, RTL-Stick, LimeMini 2.0 and AirSpy HF+ - the offset is not noticeable between them and the stop watch. You SDR is newer than mine - my be they have fixed the problem.

srs4511351, 2025-01-07:
It must be data dropouts.
I don't know if this is happening to me.
With the RPI4, gqrx does occasionally have brief pauses. I haven't noticed
it with the RPI5.
The SDRplay is 24 bit, so there is more data, but SDRglut handles it at 10
MHz! The SDRglut error messages aren't noticeably worse at 10 MHz.

----Steve