righthalfplane/SdrGlut #42: iqSDR sound dropouts

srs4511351, 2024-05-08:
Running a Raspberry Pi 4B with DietPi Debian Bookworm

This is complicated and will sound confusing. See the summary for clarification.

On a newly installed operating system, I installed speechrecognition 3.10.4, SdrGlut and iqSDR.

iqSDR had audio dropouts with about a half second of audio and a second of silence, repeating.
I found that I can repeatedly run iqSDR and it will eventually work, but that is less than half of the time.
SdrGlut is working very well with only a rare `SDRplay Dev0 RSP1A 19110C0797 (0) count 2114 sound restarted error`.

On one system where an older version (3/14/2024) of iqSDR was working, I upgraded speechrecognition to 3.10.4. [2]
After the upgrade, any new iqSDR builds had dropouts.
I downgraded speechrecognition back to the original 3.10.1 version, but any subsequent builds still had dropouts.
Copying the new iqSDR version built with speechrecognition 3.10.1 from a system where it was working had dropouts on this system. The old version still worked on this system.

**In summary,**
[System 1] new and old iqSDR versions work. It only had speechrecognition 3.10.1
[System 2] had speechrecognition 3.10.1, upgraded to speechrecognition 3.10.4, downgraded back to speechrecognition 3.10.1 works on the old version. It does not work with a working copy of the new version from [System 1]
[System 3] only had speechrecognition 3.10.4 and works on the old version compiled under speechrecognition 3.10.1. It does not work with a working copy of the new version from [System 1] or one that was compiled on this system.

It seems that any system that ever had speechrecognition 3.10.4 installed on it will run the old version, but not the new version.

The system that only had speechrecognition 3.10.1 runs both the old and new versions.

The problem seems to be partly due to speechrecognition 3.10.4 and partly the latest iqSDR version.

srs4511351, 2024-05-09:
I cannot find any instructions on how to configure options for speechrecognition.
I don't see a video on your github page about this.

To install it, I download and extract the archive from
https://pypi.org/project/SpeechRecognition/#files
then
```
cd ~/SpeechRecognition-3.10.1
sudo python3 setup.py install
```
I have RtAudio 6.0.1 installed.
I have SDRplay RSP1A, HackRF and RTL-SDR devices. I only used the SDRplay.
I never tried to use speech recognition, but I have a USB C-Media Electronics, Inc. Audio Adapter (Unitek Y-247A) Mono mic input.
I use it for input from a transceiver on digital modes like FT8.

With top, iqSDR shows about 115% CPU.
htop shows an average of 40% or less CPU usage. No individual CPU is maxed out.
I ran iqSDR many times until it worked and the CPU usage was about the same.

While playing iqSDR while it was working the dropouts started 5-10 seconds after I closed PulseAudio Control Panel while on the Input devices tab.
I was able to repeat this.

Note that the Input Devices and Output Devices menus are blank.

righthalfplane, 2024-05-09:
Which instructions did you follow for the install of iqSDR ?
Are you using the latest version of iqSDR and SdrGlut from GitHub - they have a lot of fixes in them - one of which gets the input and output device menus to appear.

Are you saying that the install of speechrecognition affects  iqSDR and SdrGlut, but you are not using speechrecognition ?

srs4511351, 2024-05-09:
I did try the latest version of iqSDR, but it was not better.
In my first post, I compared an installation done on 3/14/202 to a recent installation done after the latest version from 3/31/2024.
The the input and output device menus are blank on a recent installation of 5/7/2024. Is there a way to get the version from the application?

The installation was done like this:
```
cd ~/Desktop
git clone --depth=1 https://github.com/righthalfplane/SdrGlut.git

cd SdrGlut
make -f makefileUbuntu -j 4

cd iqSDR
make -f makefileUbuntu -j 4
```
I usually have success with makefileUbuntu on my Raspberry Pi.

I do install speechrecognition as detailed on my previous post, but I never found out how to use it.

srs4511351, 2024-05-09:
I ran a copy from another system of the 3/14/2024 installation on my newer system with speechrecognition 3.10.4.
The older version worked and the input and output menus are populated.

The installation done on 5/7/2024 on this newer system did not work and the input and output menus are blank.

srs4511351, 2024-05-10:
I decided to try building iqSDR using makefileRaspbian.
SdrGlut compiled OK.

Compiling iqSDR failed with errors related to OpenAL although it is supposed to use RtAudio.

```
src/getPalette.cpp: In function ‘int getPalette(int, char*, double*)’:
src/getPalette.cpp:1452:24: warning: ‘char* strncpy(char*, const char*, size_t)’ specified bound depends on the length of the source argument [-Wstringop-truncation]
 1452 |         if(name)strncpy(name,names[n].c_str(),strlen(names[n].c_str()));
      |                 ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
At global scope:
cc1plus: note: unrecognized command-line option ‘-Wno-return-type-c-linkage’ may have been intended to silence earlier diagnostics
g++ -std=c++11 -O2 -I/usr/include  `wx-config --cxxflags`  -I/opt/X11/include -Wall -DGL_SILENCE_DEPRECATION -Wno-return-type-c-linkage -Wno-deprecated-declarations   -c -o src/send.o src/send.cpp
g++ -std=c++11 -O2 -I/usr/include  `wx-config --cxxflags`  -I/opt/X11/include -Wall -DGL_SILENCE_DEPRECATION -Wno-return-type-c-linkage -Wno-deprecated-declarations   -c -o src/Clisten.o src/Clisten.cpp
src/send.cpp: In member function ‘int BasicPane::rxSend()’:
src/send.cpp:115:9: warning: ‘ret’ may be used uninitialized [-Wmaybe-uninitialized]
  115 |     int ret;
      |         ^~~
At global scope:
cc1plus: note: unrecognized command-line option ‘-Wno-return-type-c-linkage’ may have been intended to silence earlier diagnostics
g++  -o iqSDR.x  ./src/sdr.o ./src/mThread.o ./src/cMalloc.o ./src/iqSDR.o ./src/bStack.o ./src/sound.o ./src/getPalette.o  ./src/send.o ./src/Clisten.o -lfftw3f -lGL -lGLU -lGLEW -lsndfile  `wx-config  --libs --gl-libs` -lrtaudio -lSoapySDR -lliquid
/usr/bin/ld: ./src/sound.o: in function `stopAudio(audioInfo*) [clone .part.0]':
sound.cpp:(.text+0x18): undefined reference to `alSourceStopv'
/usr/bin/ld: sound.cpp:(.text+0x24): undefined reference to `alDeleteSources'
/usr/bin/ld: sound.cpp:(.text+0x50): undefined reference to `alDeleteBuffers'
/usr/bin/ld: sound.cpp:(.text+0x74): undefined reference to `alcGetCurrentContext'
/usr/bin/ld: sound.cpp:(.text+0x7c): undefined reference to `alcGetContextsDevice'
/usr/bin/ld: sound.cpp:(.text+0x88): undefined reference to `alcMakeContextCurrent'
/usr/bin/ld: sound.cpp:(.text+0x90): undefined reference to `alcDestroyContext'
/usr/bin/ld: sound.cpp:(.text+0x98): undefined reference to `alcCloseDevice'
/usr/bin/ld: ./src/sound.o: in function `soundClass::setBuffers(playData*)':
sound.cpp:(.text+0x2ac): undefined reference to `alBufferData'
/usr/bin/ld: sound.cpp:(.text+0x2b0): undefined reference to `alGetError'
/usr/bin/ld: sound.cpp:(.text+0x2c4): undefined reference to `alSourceQueueBuffers'
/usr/bin/ld: sound.cpp:(.text+0x2c8): undefined reference to `alGetError'
/usr/bin/ld: sound.cpp:(.text+0x31c): undefined reference to `alGetString'
/usr/bin/ld: sound.cpp:(.text+0x398): undefined reference to `alGetString'
/usr/bin/ld: ./src/sound.o: in function `soundClass::playSound()':
sound.cpp:(.text+0x424): undefined reference to `alSourcePlay'
/usr/bin/ld: sound.cpp:(.text+0x428): undefined reference to `alGetError'
/usr/bin/ld: sound.cpp:(.text+0x478): undefined reference to `alGetSourcei'
/usr/bin/ld: sound.cpp:(.text+0x48c): undefined reference to `alSourceUnqueueBuffers'
/usr/bin/ld: sound.cpp:(.text+0x4d4): undefined reference to `alGetSourcei'
/usr/bin/ld: sound.cpp:(.text+0x4f4): undefined reference to `alSourcePlay'
/usr/bin/ld: sound.cpp:(.text+0x588): undefined reference to `alGetSourcei'
/usr/bin/ld: sound.cpp:(.text+0x5c4): undefined reference to `alGetString'
/usr/bin/ld: ./src/sound.o: in function `startAudio(audioInfo*, int, int)':
sound.cpp:(.text+0x8f4): undefined reference to `alcOpenDevice'
/usr/bin/ld: sound.cpp:(.text+0x900): undefined reference to `alcCreateContext'
/usr/bin/ld: sound.cpp:(.text+0x908): undefined reference to `alcMakeContextCurrent'
/usr/bin/ld: sound.cpp:(.text+0x938): undefined reference to `alGetError'
/usr/bin/ld: sound.cpp:(.text+0x944): undefined reference to `alGenSources'
/usr/bin/ld: sound.cpp:(.text+0x948): undefined reference to `alGetError'
/usr/bin/ld: sound.cpp:(.text+0x988): undefined reference to `alGenBuffers'
/usr/bin/ld: sound.cpp:(.text+0x98c): undefined reference to `alGetError'
/usr/bin/ld: sound.cpp:(.text+0x9ac): undefined reference to `alGetString'
/usr/bin/ld: sound.cpp:(.text+0x9e0): undefined reference to `alGetString'
/usr/bin/ld: ./src/sound.o: in function `DisplayALError(unsigned char*, int)':
sound.cpp:(.text+0xd24): undefined reference to `alGetString'
collect2: error: ld returned 1 exit status
make: *** [makefileRaspbian:8: fireball] Error 1
```

righthalfplane, 2024-05-10:
Opps - I forgot that I had created a openal version of iqSDR and it got left in the default install. Reload the GitHub files to get the rtradio version. iqSDR should work better.  The  Input Devices and Output Devices menus should appear - now. I put new version number(1232) on the programs.

With the SDRplay RSP1A, don't you need to install "SDRplay_RSP_API-Linux-3.12.1.run" ?

Are you currently just using iqSDR and you wanted to send things to speechrecognition ? I would expect that the sound just needs to be piped into the speechrecognition module - what does it say when you start it - is there a "HELP" ?

srs4511351, 2024-05-11:
Nice job! It works again and the input/output menus are populated.
I see that when I select the sample rate when I start the SDR, it is not seen in the actions menu. If I select it there, then the actions menu retains the checkboxes for the sample rate and bandwidth.
IFGR does nothing when hardware AGC is unchecked. It only works when Hardware AGC is checked. That is backward.
For the spectrum display, in the left settings panel, the Maximum and Minimum values are misspelled as Maximun Value and Minimun Value.

I'm having fun checking out the radio bands in the amazing solar storm. Everything is DEAD except for ground wave. Pretty much local stations only. Very quiet
I am getting the weak ground wave from my favorite Toronto AM radio station, but no skip.
There were a few stations around 6 MHz, but I don't know why. 5 MHZ WWV is gone.

srs4511351, 2024-05-11:
Some more things I found:
File > Exit did not work. I had to close the window by clicking the X in the upper right corner.
Can not select LSB. It changes to AM. All other modes work.

Yes, I installed SDRplay_RSP_API-Linux-3.14.0.run, soon to be SDRplay_RSP_API-Linux-3.15.1.run, just released.

I mainly just want to use iqSDR and would like to experiment with speech recognition. I do not have a speech module, nor do I know any iqSDR commands, lacking any sort of documentation. What do I look for?

righthalfplane, 2024-05-11:
I was wondering why the BBC and WWV was dead last night.

The README.txt file in the iqSDR folder has a number of examples - Displaying real time TV - using dump1090 to track airplanes - saving I/Q files  etc...

Here is an example of using speech recognition to control SdrGlut. The real trick in in the program "speechcontrol.py" it does the microphone to text conversion and pipes the text to the input of a waiting program.

SdrGlut (a Sofware Defined Radio) - Testing the Voice Control Module
https://www.youtube.com/watch?v=pf-idEg8Hd8