righthalfplane/SdrGlut #44: Trouble Running SDRGlut on macOS High Sierra

TheMrGreenie, 2025-03-08:
I have an older Mac running High Sierra. XQuartz is installed. I have installed SDRGlut from the SDRGlut-1164-install.dmg file.  When I try to run SDRGlut I get the following message in the terminal window:

/Applications/SdrGlut.app/Contents/MacOS/SdrGlut3.sh ; exit;
(base) Greenes-MacBook-Pro-2:~ greenefamily$ /Applications/SdrGlut.app/Contents/MacOS/SdrGlut3.sh ; exit;
dyld: Library not loaded: /opt/X11/lib/libglut.3.dylib
  Referenced from: /Applications/SdrGlut.app/Contents/MacOS/./SdrGlut2
  Reason: Incompatible library version: SdrGlut2 requires version 13.0.0 or later, but libglut.3.dylib provides version 3.0.0
/Applications/SdrGlut.app/Contents/MacOS/SdrGlut3.sh: line 6: 61772 Abort trap: 6           ./SdrGlut2
logout
Saving session...
...copying shared history...
...saving history...truncating history files...
...completed.

Not sure how to rectify this or interpret.

righthalfplane, 2025-03-11:
Can you up grade to a newer version of XQuartz ?

I dug out my old High Sierra Machine - It has SdrGlut v1.27 on it - so that version should work on your machine. 

Building the latest version from Scratch should also work with a few tweaks.

righthalfplane, 2025-05-02:
@TheMrGreenie  The latest versions of Sdrglut and iqSDR work on my old High Sierra Machine

https://github.com/righthalfplane/SdrGlut/releases/tag/v1.33

I installed the latest version of XQuartz - that was all that was need to get them to work.