righthalfplane/SdrGlut #45: fatal error: df.h: No such file or directory

srs4511351, 2025-05-05:
On a Raspberry Pi 5, iqSDR fails during compile. I was able to get SDRglut to compile and run.
I have been successful in the past with the same procedure. What has changed in this version that could cause that?
Am I missing a requisite?

I used this procedure:
```
cd ~/Desktop
git clone --depth=1 https://github.com/righthalfplane/SdrGlut.git

cd SdrGlut
make -f makefileRaspbian -j 4

cd iqSDR
make -f makefileRaspbian -j 4
```

At this point I get the errors:

```
In file included from src/iqSDR.cpp:1:
src/iqSDR.h:33:10: fatal error: df.h: No such file or directory
   33 | #include <df.h>
      |          ^~~~~~
compilation terminated.
make: *** [<builtin>: src/iqSDR.o] Error 1
make: *** Waiting for unfinished jobs....
In file included from src/WarningBatch.h:4,
                 from src/sound.cpp:2:
src/iqSDR.h:33:10: fatal error: df.h: No such file or directory
   33 | #include <df.h>
      |          ^~~~~~
compilation terminated.
make: *** [<builtin>: src/sound.o] Error 1
```
The file df.h is located in 3 directories under ~/Desktop/SdrGlut/iqSDR

Note that in the section "Installation steps on Ubuntu 22.04", there is a typo.
git clone --depth=1 https//github.com/righthalfplane/SdrGlut.git is missing the colon.

righthalfplane, 2025-05-06:
It missed the load of the hdf libraries.

sudo apt-get install libhdf4-dev

and the -ldf needs to go on the link statement

righthalfplane, 2025-05-06:
The hdf libraries are needed for the new versions of iqSDR not for the old versions.

srs4511351, 2025-05-06:
Thanks! I thought it may have been a missed requisite, but I didn't find it in the instructions.

It works with libhdf4-dev

srs4511351, 2025-05-10:
When building iqSDR with makefileRaspbian, it fails, but if I had previously done a build with makefileUbuntu, it works.
For now, I'll just build iqSDR on my RPI with makefileUbuntu.

makefileRaspbian errors:
```
/usr/bin/ld: ./src/sweep.o: in function `writesds2dDouble(SDS2Dout*)':
sweep.cpp:(.text+0x1d58): undefined reference to `DFSDadddata'
/usr/bin/ld: sweep.cpp:(.text+0x1d64): undefined reference to `DFSDlastref'
/usr/bin/ld: sweep.cpp:(.text+0x1d7c): undefined reference to `DFANputlabel'
/usr/bin/ld: sweep.cpp:(.text+0x1ddc): undefined reference to `DFANputdesc'
/usr/bin/ld: sweep.cpp:(.text+0x1e1c): undefined reference to `DFSDclear'
/usr/bin/ld: sweep.cpp:(.text+0x1e20): undefined reference to `DFSDrestart'
/usr/bin/ld: sweep.cpp:(.text+0x1e28): undefined reference to `DFSDsetNT'
/usr/bin/ld: sweep.cpp:(.text+0x1e3c): undefined reference to `DFSDputdata'
/usr/bin/ld: ./src/sweep.o: in function `writesds3dDouble(SDS2Dout*)':
sweep.cpp:(.text+0x2088): undefined reference to `DFSDclear'
/usr/bin/ld: sweep.cpp:(.text+0x208c): undefined reference to `DFSDrestart'
/usr/bin/ld: sweep.cpp:(.text+0x2094): undefined reference to `DFSDsetNT'
/usr/bin/ld: sweep.cpp:(.text+0x20a8): undefined reference to `DFSDputdata'
/usr/bin/ld: sweep.cpp:(.text+0x20b4): undefined reference to `DFSDlastref'
/usr/bin/ld: sweep.cpp:(.text+0x20cc): undefined reference to `DFANputlabel'
/usr/bin/ld: sweep.cpp:(.text+0x212c): undefined reference to `DFANputdesc'
/usr/bin/ld: sweep.cpp:(.text+0x21b0): undefined reference to `DFSDadddata'
/usr/bin/ld: ./src/sweep.o: in function `writesds2dFloat(SDS2Dout*)':
sweep.cpp:(.text+0x23e0): undefined reference to `DFSDadddata'
/usr/bin/ld: sweep.cpp:(.text+0x23e8): undefined reference to `DFSDlastref'
/usr/bin/ld: sweep.cpp:(.text+0x2400): undefined reference to `DFANputlabel'
/usr/bin/ld: sweep.cpp:(.text+0x2460): undefined reference to `DFANputdesc'
/usr/bin/ld: sweep.cpp:(.text+0x24f8): undefined reference to `DFSDclear'
/usr/bin/ld: sweep.cpp:(.text+0x24fc): undefined reference to `DFSDrestart'
/usr/bin/ld: sweep.cpp:(.text+0x2504): undefined reference to `DFSDsetNT'
/usr/bin/ld: sweep.cpp:(.text+0x2518): undefined reference to `DFSDputdata'
/usr/bin/ld: ./src/sweep.o: in function `writesds3dFloat(SDS2Dout*)':
sweep.cpp:(.text+0x2768): undefined reference to `DFSDadddata'
/usr/bin/ld: sweep.cpp:(.text+0x2774): undefined reference to `DFSDlastref'
/usr/bin/ld: sweep.cpp:(.text+0x278c): undefined reference to `DFANputlabel'
/usr/bin/ld: sweep.cpp:(.text+0x27f0): undefined reference to `DFANputdesc'
/usr/bin/ld: sweep.cpp:(.text+0x287c): undefined reference to `DFSDclear'
/usr/bin/ld: sweep.cpp:(.text+0x2880): undefined reference to `DFSDrestart'
/usr/bin/ld: sweep.cpp:(.text+0x2888): undefined reference to `DFSDsetNT'
/usr/bin/ld: sweep.cpp:(.text+0x289c): undefined reference to `DFSDputdata'
/usr/bin/ld: ./src/sweep.o: in function `Sweep::openSweepFile(char*)':
sweep.cpp:(.text+0x5324): undefined reference to `DFSDclear'
/usr/bin/ld: sweep.cpp:(.text+0x5328): undefined reference to `DFSDrestart'
/usr/bin/ld: sweep.cpp:(.text+0x5340): undefined reference to `DFSDgetdims'
/usr/bin/ld: sweep.cpp:(.text+0x53c0): undefined reference to `DFSDgetdata'
/usr/bin/ld: sweep.cpp:(.text+0x5480): undefined reference to `DFSDlastref'
/usr/bin/ld: sweep.cpp:(.text+0x549c): undefined reference to `DFANgetlabel'
/usr/bin/ld: sweep.cpp:(.text+0x54c8): undefined reference to `DFANgetdesc'
/usr/bin/ld: ./src/sweep.o: in function `writesds(SDS2Dout*)':
sweep.cpp:(.text+0xae64): undefined reference to `DFR8restart'
/usr/bin/ld: sweep.cpp:(.text+0xae7c): undefined reference to `DFR8putimage'
/usr/bin/ld: sweep.cpp:(.text+0xae88): undefined reference to `DFR8lastref'
/usr/bin/ld: sweep.cpp:(.text+0xaea0): undefined reference to `DFANputlabel'
/usr/bin/ld: sweep.cpp:(.text+0xaf04): undefined reference to `DFANputdesc'
/usr/bin/ld: sweep.cpp:(.text+0xaf94): undefined reference to `DFR8addimage'
collect2: error: ld returned 1 exit status
make: *** [makefileRaspbian:11: fireball] Error 1

```

righthalfplane, 2025-05-11:
It the same problem, almost, just add -ldf or -lhdf to the link line.